xiaoqingfengATGH/HomeLede #28: 系统日志中一直有网卡错误出现

lam640, 2020-07-10:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
系统日志提示报错
Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue
无
## 2. 详细叙述

### (1) 具体问题

A：系统日志中大量类似错误
 kern.warn kernel: [121669.543660] ixgbe 0000:01:00.1: Warning firmware error detected FWSM: 0x00000000
### (2) 路由器型号和固件版本

主机名 | OpenWrt
-- | --
主机型号 | Intel(R) Xeon(R) CPU E3-1265L v3 @ 2.50GHz : 4 Core 8 Thread (CpuMark : 122277.416890 Scores)
CPU 信息 | 1502.398 MHz +52.0°C (high = +84.0°C, crit = +90.0°C)
固件版本 | HomeLede v2020.07.07 based on OpenWrt R20.7.7 / LuCI Master (git-20.187.45844-a7450cd)
内核版本 | 5.4.50
本地时间 | Fri Jul 10 13:53:49 2020
运行时间 | 1d 9h 49m 20s
平均负载 | 0.50, 0.18, 0.06
CPU 使用率（%）

A：

### (3) 详细日志

A：[121766.305260] ixgbe 0000:01:00.1: Warning firmware error detected FWSM: 0x00000000
[121766.529255] ixgbe 0000:01:00.0: Warning firmware error detected FWSM: 0x00000000
[121768.325211] ixgbe 0000:01:00.1: Warning firmware error detected FWSM: 0x00000000
[121768.545230] ixgbe 0000:01:00.0: Warning firmware error detected FWSM: 0x00000000

xiaoqingfengATGH, 2020-07-24:
网卡驱动问题，确认下你的网卡型号。

lam640, 2020-07-24:
> 网卡驱动问题，确认下你的网卡型号。
我的网卡是intel x520-t2，我在intel官网查询到ixgbe驱动在5.6版本上面才支持5的内核，不知是不是因为这个导致错误，还望大神能帮忙看下，感谢

xiaoqingfengATGH, 2020-07-27:
貌似目前源码中没有这款网卡驱动。

lam640, 2020-07-27:
> 貌似目前源码中没有这款网卡驱动。

应该和82599是同一款驱动的

closed 2020-09-21 by xiaoqingfengATGH