xiaoqingfengATGH/HomeLede #29: 0711作为旁路时 设备无法上网

KyonCN, 2020-07-16:
主路由时ikuai 能够ping通114.114.114.114 和.baidu
旁路由homelede 能够ping通114.114.114.114 dns解析正常 可在线安装插件 默认状态 未开任何其他功能 
主旁路由能正常互相ping通 所有设置都参照Google网盘里都二级路由示意图且homelede所有防火墙都被设置为允许
内网设备能够正常获取ip 网关 dns 且能够正常解析网址ip  nslookup返回的dns服务器地址就是旁路由ip
内网设备无法访问外网 只能ping通主旁路由  无法ping通任何外网ip
tranceroute追踪显示到homelede处就没有响应了 

`
Software-Version
-------------------------------------------------
OpenWrt - HomeLede  v2020.07.11 based on OpenWrt R20.7.7
LuCI - git-20.191.45716-b73d3a4

Output of "ip a show"
-------------------------------------------------
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host 
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc fq_codel master br-lan state UP group default qlen 1000
    link/ether be:eb:d3:69:26:73 brd ff:ff:ff:ff:ff:ff
3: ip6tnl0@NONE: <NOARP> mtu 1452 qdisc noop state DOWN group default qlen 1000
    link/tunnel6 :: brd ::
4: sit0@NONE: <NOARP> mtu 1480 qdisc noop state DOWN group default qlen 1000
    link/sit 0.0.0.0 brd 0.0.0.0
5: ip_vti0@NONE: <NOARP> mtu 1480 qdisc noop state DOWN group default qlen 1000
    link/ipip 0.0.0.0 brd 0.0.0.0
6: ip6_vti0@NONE: <NOARP> mtu 1332 qdisc noop state DOWN group default qlen 1000
    link/tunnel6 :: brd ::
7: dummy0: <BROADCAST,NOARP> mtu 1500 qdisc noop state DOWN group default qlen 1000
    link/ether d6:9a:33:96:32:7b brd ff:ff:ff:ff:ff:ff
8: gre0@NONE: <NOARP> mtu 1476 qdisc noop state DOWN group default qlen 1000
    link/gre 0.0.0.0 brd 0.0.0.0
9: gretap0@NONE: <BROADCAST,MULTICAST> mtu 1462 qdisc noop state DOWN group default qlen 1000
    link/ether 00:00:00:00:00:00 brd ff:ff:ff:ff:ff:ff
10: erspan0@NONE: <BROADCAST,MULTICAST> mtu 1450 qdisc noop state DOWN group default qlen 1000
    link/ether 00:00:00:00:00:00 brd ff:ff:ff:ff:ff:ff
11: bond0: <BROADCAST,MULTICAST,MASTER> mtu 1500 qdisc noop state DOWN group default qlen 1000
    link/ether 52:63:b3:cf:9c:19 brd ff:ff:ff:ff:ff:ff
12: br-lan: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP group default qlen 1000
    link/ether be:eb:d3:69:26:73 brd ff:ff:ff:ff:ff:ff
    inet 192.168.2.201/24 brd 192.168.2.255 scope global br-lan
       valid_lft forever preferred_lft forever
    inet6 fd4d:a9f:58e3::1/60 scope global noprefixroute 
       valid_lft forever preferred_lft forever
    inet6 fe80::bceb:d3ff:fe69:2673/64 scope link 
       valid_lft forever preferred_lft forever
18: docker0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc noqueue state DOWN group default 
    link/ether 02:42:fc:2f:8a:01 brd ff:ff:ff:ff:ff:ff
    inet 172.17.0.1/16 brd 172.17.255.255 scope global docker0
       valid_lft forever preferred_lft forever

Output of "ip route show"
-------------------------------------------------
default via 192.168.2.200 dev br-lan proto static 
172.17.0.0/16 dev docker0 proto kernel scope link src 172.17.0.1 linkdown 
192.168.2.0/24 dev br-lan proto kernel scope link src 192.168.2.201

Output of "ip rule show"
-------------------------------------------------
0:	from all lookup local
32766:	from all lookup main
32767:	from all lookup default

Output of "ip route list table 1-250"
-------------------------------------------------

Output of "iptables -L -t mangle -v -n"
-------------------------------------------------
Chain PREROUTING (policy ACCEPT 6063 packets, 390K bytes)
 pkts bytes target     prot opt in     out     source               destination         

Chain INPUT (policy ACCEPT 1088 packets, 106K bytes)
 pkts bytes target     prot opt in     out     source               destination         
 1092  106K RRDIPT_INPUT  all  --  *      *       0.0.0.0/0            0.0.0.0/0           

Chain FORWARD (policy ACCEPT 2169 packets, 137K bytes)
 pkts bytes target     prot opt in     out     source               destination         
 2181  138K RRDIPT_FORWARD  all  --  *      *       0.0.0.0/0            0.0.0.0/0           
 2152  136K TCPMSS     tcp  --  *      br-lan  0.0.0.0/0            0.0.0.0/0            tcp flags:0x06/0x02 /* !fw3: Zone newzone MTU fixing */ TCPMSS clamp to PMTU

Chain OUTPUT (policy ACCEPT 1176 packets, 442K bytes)
 pkts bytes target     prot opt in     out     source               destination         
 1180  442K RRDIPT_OUTPUT  all  --  *      *       0.0.0.0/0            0.0.0.0/0           

Chain POSTROUTING (policy ACCEPT 3347 packets, 580K bytes)
 pkts bytes target     prot opt in     out     source               destination         

Chain RRDIPT_FORWARD (1 references)
 pkts bytes target     prot opt in     out     source               destination         
  100  4816 RETURN     all  --  *      *       192.168.2.23         0.0.0.0/0           
    0     0 RETURN     all  --  *      *       0.0.0.0/0            192.168.2.23        
 2079  133K RETURN     all  --  *      *       192.168.2.102        0.0.0.0/0           
    0     0 RETURN     all  --  *      *       0.0.0.0/0            192.168.2.102       
    0     0 RETURN     all  --  *      *       192.168.2.200        0.0.0.0/0           
    0     0 RETURN     all  --  *      *       0.0.0.0/0            192.168.2.200       

Chain RRDIPT_INPUT (1 references)
 pkts bytes target     prot opt in     out     source               destination         
    0     0 RETURN     all  --  eth0   *       0.0.0.0/0            0.0.0.0/0           
  956 91580 RETURN     all  --  br-lan *       0.0.0.0/0            0.0.0.0/0           

Chain RRDIPT_OUTPUT (1 references)
 pkts bytes target     prot opt in     out     source               destination         
    0     0 RETURN     all  --  *      eth0    0.0.0.0/0            0.0.0.0/0           
 1044  428K RETURN     all  --  *      br-lan  0.0.0.0/0            0.0.0.0/0           

Chain qos_Default (0 references)
 pkts bytes target     prot opt in     out     source               destination         
    0     0 CONNMARK   all  --  *      *       0.0.0.0/0            0.0.0.0/0            CONNMARK restore mask 0xf
    0     0 qos_Default_ct  all  --  *      *       0.0.0.0/0            0.0.0.0/0            mark match 0x0/0xf
    0     0 MARK       udp  --  *      *       0.0.0.0/0            0.0.0.0/0            mark match 0x0/0xf0 length 0:500 MARK xset 0x22/0xff
    0     0 MARK       icmp --  *      *       0.0.0.0/0            0.0.0.0/0            MARK xset 0x11/0xff
    0     0 MARK       tcp  --  *      *       0.0.0.0/0            0.0.0.0/0            mark match 0x0/0xf0 tcp spts:1024:65535 dpts:1024:65535 MARK xset 0x44/0xff
    0     0 MARK       udp  --  *      *       0.0.0.0/0            0.0.0.0/0            mark match 0x0/0xf0 udp spts:1024:65535 dpts:1024:65535 MARK xset 0x44/0xff
    0     0 CONNMARK   all  --  *      *       0.0.0.0/0            0.0.0.0/0            CONNMARK save mask 0xff

Chain qos_Default_ct (1 references)
 pkts bytes target     prot opt in     out     source               destination         
    0     0 MARK       tcp  --  *      *       0.0.0.0/0            0.0.0.0/0            mark match 0x0/0xf tcp multiport ports 22,53 /* ssh, dns */ MARK xset 0x11/0xff
    0     0 MARK       udp  --  *      *       0.0.0.0/0            0.0.0.0/0            mark match 0x0/0xf udp multiport ports 22,53 /* ssh, dns */ MARK xset 0x11/0xff
    0     0 MARK       tcp  --  *      *       0.0.0.0/0            0.0.0.0/0            mark match 0x0/0xf tcp multiport ports 20,21,25,80,110,443,993,995 /* ftp, smtp, http(s), imap */ MARK xset 0x33/0xff
    0     0 MARK       tcp  --  *      *       0.0.0.0/0            0.0.0.0/0            mark match 0x0/0xf tcp multiport ports 5190 /* AOL, iChat, ICQ */ MARK xset 0x22/0xff
    0     0 MARK       udp  --  *      *       0.0.0.0/0            0.0.0.0/0            mark match 0x0/0xf udp multiport ports 5190 /* AOL, iChat, ICQ */ MARK xset 0x22/0xff
    0     0 CONNMARK   all  --  *      *       0.0.0.0/0            0.0.0.0/0            CONNMARK save mask 0xff`

KyonCN, 2020-07-17:
解决了  要把LAN口的链路桥接取消勾选

closed 2020-07-17 by KyonCN

xiaoqingfengATGH, 2020-07-18:
看来方案里应该强调下这个细节。

xiaoqingfengATGH, 2020-07-18:
谢谢反馈。