xiaoqingfengATGH/HomeLede #31: 718版本DNS有问题，总是会很快无法解析

chen8148, 2020-07-23:
718版本DNS有问题，在我使用中，DNS总是会很快无法解析。
而在这期间我只是配置了homeclash的订阅地址，并按照clash的要求在lan中关闭了ipv6的DHCP。
之后突然发现不能打开网页了，测试DNS超时。试着关闭去广告、smartDNS和ChinaDNS都没用。

KyonCN, 2020-07-23:
你用最新版本的代码编译 我试了k5最后commit后的固件 没啥问题

manny0, 2020-07-27:
我也遇到过这个问题，发现是设备端获取了IP v6的DNS服务器地址，手动删除以后好了。关闭LAN的IPv6管理再重新获取IP，貌似DNS服务器就只会获取到IPv4的了，设备也可以正常解析IP地址了

xiaoqingfengATGH, 2020-07-27:
建议关闭WAN的，“使用内置ipv6”管理。

chen8148, 2020-08-19:
> 建议关闭WAN的，“使用内置ipv6”管理。

我应该关闭wan的使用内置ipv6”管理，还是LAN的？这两个中都有这个设置项。

chen8148, 2020-08-19:
> 建议关闭WAN的，“使用内置ipv6”管理。

另外，这个问题是否在之后的版本中修复了？因为我想以后可以使用ipv6的公网ip。

xiaoqingfengATGH, 2020-08-25:
请尝试固件最新代码。

近期Passwall修改了dns设置，固件里也做了对应调整。

gulinghua, 2020-08-31:
> > 建议关闭WAN的，“使用内置ipv6”管理。
> 
> 我应该关闭wan的使用内置ipv6”管理，还是LAN的？这两个中都有这个设置项。
> 另外，这个问题是否在之后的版本中修复了？因为我想以后可以使用ipv6的公网ip。

我单关wan没起作用，两个都关了以后就没遇到问题了

closed 2020-11-16 by xiaoqingfengATGH