xiaoqingfengATGH/HomeLede #32: overlay扩容无法挂载

newageworld, 2020-07-23:
对overlay扩容后，一直无法实现挂载，具体情况见截图。不知道是怎么回事？

KyonCN, 2020-07-23:
你把overlay下所有文件复制到sdb1里去  然后把sdb1改名成overlay  最后删除根目录下的overlay  把/mnt/overlay 软连接到根目录下就可以里

xiaoqingfengATGH, 2020-07-27:
目前版本沿用的官方代码fstab似乎有问题。/etc/config/fstab会读取失败。

v26cn, 2020-07-27:
同样的问题,手动挂载可以显示,重启就失效了,fstab没发现问题,折腾两天了

xiaoqingfengATGH, 2020-07-28:
已经反馈Lean团队，看看是否有办法解决。

官方snapshot可能没有问题，但目前编译有错，随后再试。

glen-1, 2020-08-07:
同样的问题，
如果可以介绍一个编译时更改/overlay分区大小的方法就更好了

Yiipanny, 2020-08-09:
> 同样的问题，
> 如果可以介绍一个编译时更改/overlay分区大小的方法就更好了

在make menuconfig的时候，调TARGET_ROOTFS_PARTSIZE这个参数，单位是MB

ghell, 2020-08-24:
> 已经反馈Lean团队，看看是否有办法解决。
> 
> 官方snapshot可能没有问题，但目前编译有错，随后再试。

作者大大能否编译时把overlay稍微搞大一点？ 100M实在太小了，现在磁盘空间应该不是大问题。谢谢

aozaki-kuro, 2020-08-28:
0815版本同样遇到了这个问题。
尝试建立sda3，复制文件后在选择挂载为外部overlay，重启后发现操作未生效，同样操作步骤在0620成功。

如果短期内无法解决希望下一个版本可以编译时把overlay调整大一些（例如1-2G）😢

pauloy007, 2020-09-08:
已有解决办法，具体请到HomeLede 2群 咨询

newageworld, 2020-09-08:
[image]

newageworld, 2020-09-08:
根据楼上的操作一次成功，给你点赞

xiaole51, 2020-09-15:
今天遇到这问题,重装了两次,复位了两次.
以为我哪里操作错了....
最后无奈谷歌后找到这个贴. 这坑啊....

wylyeak, 2020-11-01:
> 根据楼上的操作一次成功，给你点赞

之前也这么操作过， 但是实际各应用的文件还是写到原分区了。  这个扩容的感觉是不生效的。

wcwhqs, 2023-08-31:
> 

对的，这个方法没法overlayfs:/overlay扩容