xiaoqingfengATGH/HomeLede #38: 请问使用ipv6的话该如何设置

dovela, 2020-08-25:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：PVE环境下完全新装的718版固件，“禁止解析ipv6 DNS记录”关掉了，终端可以获取到2409:8a3c开头的ipv6公网地址，但是实际不能访问纯ipv6网站，是有其他地方还需要设置吗？

### (2) 路由器型号和固件版本

A：i5软路由

### (3) 详细日志

A：

xiaoqingfengATGH, 2020-08-26:
先确认你本地isp ipv6可以正常用。

固件4个dns组件，把ipv6 dns解析打开。

其他的应该不需要什么了，具体电报群中讨论吧。

xiaoxinpro, 2020-09-01:
@xiaoqingfengATGH 

我的现象一样，路由Wan和Lan都有ipv6地址，

禁止解析IPV6 DNS的开关也关上了

但就是不能访问IPV6

我使用原版的OpenWrt默认访问IPV6一切正常

xiaoxinpro, 2020-09-07:
### @dovela 我找到访问纯ipv6的办法了：

1. 首先还是把所有DNS服务的禁止解析ipv6的开关关闭

2. 再在 `网络 - 负载均衡 - 接口` 界面禁用全部接口，`网络 - 负载均衡 - 策略` 修改备用成员为`默认（使用主路由表）`

3. 最后重启你的OpenWrt和你的客户端，访问 http://www.test-ipv6.com/ 测试通过！

### @xiaoqingfengATGH 给大大两个建议：

1. 建议默认启用DNS的IPV6的解析，毕竟IPV6的优点太多了，访问低延时（速度快）、带宽费用低。。。

2. 建议默认关闭多拨和负载均衡，毕竟现在运营商越来越多的不支持多拨了，默认开多拨很有可能导致小白宽带账号被封（然后打运营商电话恢复宽带账号很麻烦）。

xiaoqingfengATGH, 2020-09-11:
感谢细心测试及反馈。

从群里调查的情况看，国内一些地区v6支持相对完善，开启后可以提升上网体验。

但还有一些地区（相对较多），开启v6后会有各种各样问题。

因此，固件会暂时保持观望，未来会渐渐演变到默认v6。

closed 2020-09-11 by xiaoqingfengATGH