xiaoqingfengATGH/HomeLede #41: dns求教

gulinghua, 2020-08-31:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：我用的是0728的版本，发现解析国内的网址速度很慢，经常发现解析失败
用dig命令看5053端口已经解析到地址了，但是54端口还是没有结果，看上去像是dnsmasq的问题
想了解一下，chinadns-ng是如何重定向到dnsmasq的？ 我遇到的这个问题，可能的原因是什么

谢谢

### (2) 路由器型号和固件版本

A：0728版本, 软路由

### (3) 详细日志

A：
root@OpenWrt:/tmp# dig ip111.cn @127.0.0.1 -p 5053; dig ip111.cn @127.0.0.1 -p 54

; <<>> DiG 9.16.4 <<>> ip111.cn @127.0.0.1 -p 5053
;; global options: +cmd
;; Got answer:
;; ->>HEADER<<- opcode: QUERY, status: NOERROR, id: 63741
;; flags: qr rd ra; QUERY: 1, ANSWER: 1, AUTHORITY: 0, ADDITIONAL: 0

;; QUESTION SECTION:
;ip111.cn.			IN	A

;; ANSWER SECTION:
ip111.cn.		3584	IN	A	121.41.110.161

;; Query time: 0 msec
;; SERVER: 127.0.0.1#5053(127.0.0.1)
;; WHEN: Mon Aug 31 23:51:01 CST 2020
;; MSG SIZE  rcvd: 50

; <<>> DiG 9.16.4 <<>> ip111.cn @127.0.0.1 -p 54
;; global options: +cmd
;; Got answer:
;; ->>HEADER<<- opcode: QUERY, status: SERVFAIL, id: 24416
;; flags: qr rd ra; QUERY: 1, ANSWER: 0, AUTHORITY: 0, ADDITIONAL: 1

;; OPT PSEUDOSECTION:
; EDNS: version: 0, flags:; udp: 4096
;; QUESTION SECTION:
;ip111.cn.			IN	A

;; Query time: 16 msec
;; SERVER: 127.0.0.1#54(127.0.0.1)
;; WHEN: Mon Aug 31 23:51:01 CST 2020
;; MSG SIZE  rcvd: 37

gulinghua, 2020-08-31:
搜了一下别人的反馈，貌似0718遇到了同样的问题

gulinghua, 2020-08-31:
学习了一下别人的comment，把ipv6彻底关了以后没有复现问题

gulinghua, 2020-09-02:
发现问题还有，更新了0815的测试版，仍然遇到问题
总结一下现象：
1. 出问题的都是国内地址
2. 从nslookup结果看，127.0.0.1#54 返回SERVFAIL
3. 手动测试 127.0.0.1#5053和#6053，都是秒回结果

xiaoqingfengATGH, 2020-09-11:
测试下最新09.05正式版本？

sekaiacg, 2020-09-14:
@xiaoqingfengATGH 用9.12 test的代码编译的版本同样问题，baidu,bilibili这些常用的网站打不开或者很慢
1.adg关闭也不行
2.将ipv6管理关闭也不行，不是ipv6的问题
3.将终端设备的dns改为114没问题了

gulinghua, 2020-09-15:
9.5正式版用了好多天了，问题依旧

sekaiacg, 2020-09-15:
有人说是dnsmasq的问题
我进行了如下测试：将adg的上游dns由127.0.0.1:54改成127.0.0.1:5053
跳过这个东西，竟然没有问题了，但是不知道会不会引起其他问题

gulinghua, 2020-09-15:
试了下关闭 dnsmasq-china-list，好像没啥问题了
dnsmasq-china-list和chinadns-ng两个的作用完全重复了 @xiaoqingfengATGH

ahjsrhj, 2020-09-18:
有同样的问题，经常性的 dns 解析不可用

xiaoqingfengATGH, 2020-09-20:
如果遇到了卸载dnsmasq-chinalist使用吧。具体我再观察下。

xiaoqingfengATGH, 2020-09-21:
修改dnsmasq-china-list内置国内域名列表默认dns指向固件国内组dns（127.0.0.1#6053），请大家测试观测状态。

btdwv, 2020-10-12:
安装10月1日的测试版之后，这个问题没再出现过

liyaodong, 2020-10-27:
10月1日的测试版安装后问题依旧

尝试 关闭 ipv6 相关的东西无果，准备试试 卸载dnsmasq-chinalist

liyaodong, 2020-10-27:
还是没用 把 adg 的 upstream DNS 改成了 7053，这样国外能上网，国内慢点。但至少可用。

附 dig 结果，可以看到 54 根本没相应。重启过 N 此也删掉了 dnsmasq-chinalist，关掉了 ipv6 相关的设置。

```
; <<>> DiG 9.16.4 <<>> ip111.cn @127.0.0.1 -p 54
;; global options: +cmd
;; connection timed out; no servers could be reached

; <<>> DiG 9.16.4 <<>> ip111.cn @127.0.0.1 -p 6053
;; global options: +cmd
;; Got answer:
;; ->>HEADER<<- opcode: QUERY, status: NOERROR, id: 15828
;; flags: qr rd ra; QUERY: 1, ANSWER: 1, AUTHORITY: 0, ADDITIONAL: 0

;; QUESTION SECTION:
;ip111.cn.			IN	A

;; ANSWER SECTION:
ip111.cn.		3600	IN	A	121.41.110.161

;; Query time: 56 msec
;; SERVER: 127.0.0.1#6053(127.0.0.1)
;; WHEN: Tue Oct 27 16:38:29 CST 2020
;; MSG SIZE  rcvd: 50

; <<>> DiG 9.16.4 <<>> ip111.cn @127.0.0.1 -p 7053
;; global options: +cmd
;; Got answer:
;; ->>HEADER<<- opcode: QUERY, status: NOERROR, id: 39494
;; flags: qr rd ra; QUERY: 1, ANSWER: 1, AUTHORITY: 0, ADDITIONAL: 1

;; OPT PSEUDOSECTION:
; EDNS: version: 0, flags:; udp: 1232
;; QUESTION SECTION:
;ip111.cn.			IN	A

;; ANSWER SECTION:
ip111.cn.		2399	IN	A	121.41.110.161

;; Query time: 192 msec
;; SERVER: 127.0.0.1#7053(127.0.0.1)
;; WHEN: Tue Oct 27 16:38:30 CST 2020
;; MSG SIZE  rcvd: 53
```

xiaoqingfengATGH, 2020-10-29:
如果dnsmasq失效，大部分dns解析都会失效。

你遇到的情况十分罕见。我试了很多情况还是没有重现。

liyaodong, 2020-10-30:
如果有这个问题的检查一下 网络 - DHCP/DNS 中 DNS 转发的设置，我改成 5053（chinadns-ng） 就好了，但是还没理解 clash 的关闭和开启对 dnsmasq 的转发有什么影响，dnsmasq 和 clash 的 Fake IP 怎样沟通工作的。

liyaodong, 2020-10-30:
clash 启动的时候会把 dnsmasq DNS 转发设置成 7874，也就是 clash 自己的端口。 clash 关闭时会清空这个 dnsmasq DNS转发。

xiaoqingfengATGH, 2020-10-30:
DNS方案固件WIKI有说明，可以参考。

另外，如果使用passwall，试试dns页面的 清空ipset

ugvfpdcuwfnh, 2020-12-01:
54端口确实没有反映。
我的办法是在adguard里面把54换成5053，fakeip相当于被跳过去了。

1261685372, 2021-03-05:
把DHCP/DNS DNS转发改为6063端口成功解决无法解析的问题  这样子相当于跳过了chinadns-ng
adguard 上游dns改为5053无效 个人感觉是chinadns-ng存在bug引起的
系统版本：20.09.05正式版

1261685372, 2021-03-06:
感觉是chinadns-ng挂掉了 于是在启动项中重启chinadns-ng 出现以下系统日志 
Sat Mar  6 14:57:26 2021 daemon.err uhttpd[5295]: sh: /etc/init.d/dockerman: line 6: config_load: not found
Sat Mar  6 14:57:26 2021 daemon.err uhttpd[5295]: sh: /etc/init.d/dockerman: line 7: config_get: not found
Sat Mar  6 14:57:26 2021 daemon.err uhttpd[5295]: sh: /etc/init.d/homeredirect: line 155: syntax error: bad for loop variable
Sat Mar  6 14:57:27 2021 daemon.err uhttpd[5295]: BusyBox v1.31.1 () multi-call binary.
Sat Mar  6 14:57:27 2021 daemon.err uhttpd[5295]:
Sat Mar  6 14:57:27 2021 daemon.err uhttpd[5295]: Usage: basename FILE [SUFFIX]
Sat Mar  6 14:57:27 2021 daemon.err uhttpd[5295]:
Sat Mar  6 14:57:27 2021 daemon.err uhttpd[5295]: Strip directory path and .SUFFIX from FILE
Sat Mar  6 14:57:27 2021 daemon.err uhttpd[5295]: BusyBox v1.31.1 () multi-call binary.
Sat Mar  6 14:57:27 2021 daemon.err uhttpd[5295]:
Sat Mar  6 14:57:27 2021 daemon.err uhttpd[5295]: Usage: basename FILE [SUFFIX]
Sat Mar  6 14:57:27 2021 daemon.err uhttpd[5295]:
Sat Mar  6 14:57:27 2021 daemon.err uhttpd[5295]: Strip directory path and .SUFFIX from FILE
Sat Mar  6 14:57:27 2021 daemon.err uhttpd[5295]: BusyBox v1.31.1 () multi-call binary.
Sat Mar  6 14:57:27 2021 daemon.err uhttpd[5295]:
Sat Mar  6 14:57:27 2021 daemon.err uhttpd[5295]: Usage: basename FILE [SUFFIX]
Sat Mar  6 14:57:27 2021 daemon.err uhttpd[5295]:
Sat Mar  6 14:57:27 2021 daemon.err uhttpd[5295]: Strip directory path and .SUFFIX from FILE
Sat Mar  6 14:57:27 2021 daemon.err uhttpd[5295]: BusyBox v1.31.1 () multi-call binary.
Sat Mar  6 14:57:27 2021 daemon.err uhttpd[5295]:
Sat Mar  6 14:57:27 2021 daemon.err uhttpd[5295]: Usage: basename FILE [SUFFIX]
Sat Mar  6 14:57:27 2021 daemon.err uhttpd[5295]:
Sat Mar  6 14:57:27 2021 daemon.err uhttpd[5295]: Strip directory path and .SUFFIX from FILE
Sat Mar  6 14:57:27 2021 daemon.err uhttpd[5295]: BusyBox v1.31.1 () multi-call binary.
Sat Mar  6 14:57:27 2021 daemon.err uhttpd[5295]:
Sat Mar  6 14:57:27 2021 daemon.err uhttpd[5295]: Usage: basename FILE [SUFFIX]
Sat Mar  6 14:57:27 2021 daemon.err uhttpd[5295]:
Sat Mar  6 14:57:27 2021 daemon.err uhttpd[5295]: Strip directory path and .SUFFIX from FILE
Sat Mar  6 14:57:37 2021 daemon.err uhttpd[5295]: sh: /etc/init.d/dockerman: line 6: config_load: not found
Sat Mar  6 14:57:37 2021 daemon.err uhttpd[5295]: sh: /etc/init.d/dockerman: line 7: config_get: not found
Sat Mar  6 14:57:38 2021 daemon.err uhttpd[5295]: sh: /etc/init.d/homeredirect: line 155: syntax error: bad for loop variable
Sat Mar  6 14:57:38 2021 daemon.err uhttpd[5295]: BusyBox v1.31.1 () multi-call binary.
Sat Mar  6 14:57:38 2021 daemon.err uhttpd[5295]:
Sat Mar  6 14:57:38 2021 daemon.err uhttpd[5295]: Usage: basename FILE [SUFFIX]
Sat Mar  6 14:57:38 2021 daemon.err uhttpd[5295]:
Sat Mar  6 14:57:38 2021 daemon.err uhttpd[5295]: Strip directory path and .SUFFIX from FILE
Sat Mar  6 14:57:38 2021 daemon.err uhttpd[5295]: BusyBox v1.31.1 () multi-call binary.
Sat Mar  6 14:57:38 2021 daemon.err uhttpd[5295]:
Sat Mar  6 14:57:38 2021 daemon.err uhttpd[5295]: Usage: basename FILE [SUFFIX]
Sat Mar  6 14:57:38 2021 daemon.err uhttpd[5295]:
Sat Mar  6 14:57:38 2021 daemon.err uhttpd[5295]: Strip directory path and .SUFFIX from FILE
Sat Mar  6 14:57:38 2021 daemon.err uhttpd[5295]: BusyBox v1.31.1 () multi-call binary.
Sat Mar  6 14:57:38 2021 daemon.err uhttpd[5295]:
Sat Mar  6 14:57:38 2021 daemon.err uhttpd[5295]: Usage: basename FILE [SUFFIX]
Sat Mar  6 14:57:38 2021 daemon.err uhttpd[5295]:
Sat Mar  6 14:57:38 2021 daemon.err uhttpd[5295]: Strip directory path and .SUFFIX from FILE
Sat Mar  6 14:57:38 2021 daemon.err uhttpd[5295]: BusyBox v1.31.1 () multi-call binary.
Sat Mar  6 14:57:38 2021 daemon.err uhttpd[5295]:
Sat Mar  6 14:57:38 2021 daemon.err uhttpd[5295]: Usage: basename FILE [SUFFIX]
Sat Mar  6 14:57:38 2021 daemon.err uhttpd[5295]:
Sat Mar  6 14:57:38 2021 daemon.err uhttpd[5295]: Strip directory path and .SUFFIX from FILE
Sat Mar  6 14:57:38 2021 daemon.err uhttpd[5295]: BusyBox v1.31.1 () multi-call binary.
Sat Mar  6 14:57:38 2021 daemon.err uhttpd[5295]:
Sat Mar  6 14:57:38 2021 daemon.err uhttpd[5295]: Usage: basename FILE [SUFFIX]
Sat Mar  6 14:57:38 2021 daemon.err uhttpd[5295]:
Sat Mar  6 14:57:38 2021 daemon.err uhttpd[5295]: Strip directory path and .SUFFIX from FILE

1261685372, 2021-03-06:
> 把DHCP/DNS DNS转发改为6053端口成功解决无法解析的问题 这样子相当于跳过了chinadns-ng
> adguard 上游dns改为5053无效 个人感觉是chinadns-ng存在bug引起的
> 系统版本：20.09.05正式版

又用了2021.02.27版本试了一下 一样的问题   
以百度为例（国内网站）
54 5053 6053端口都能解析  唯独7053无法解析
以github为例（国外网站）
54 5053 7053无法解析 6053端口解析正常

个人怀疑是dnscryt-proxy的环节存在问题 
系统日志显示dnscrypt解析超时
Sat Mar  6 19:36:00 2021 daemon.err dnscrypt-proxy[26091]: [2021-03-06 11:36:00] [ERROR] Get "https://doh.dnslify.com/dns-query?dns=yv4BAAABAAAAAAABAAACAAEAACkQAAAAAAAAFAAMABDbp-l0iKNgH444KK6-LOpm": net/http: request canceled while waiting for connection (Client.Timeout exceeded while awaiting headers)
Sat Mar  6 19:36:00 2021 daemon.err dnscrypt-proxy[26091]: [2021-03-06 11:36:00] [NOTICE] dnscrypt-proxy is waiting for at least one server to be reachable