xiaoqingfengATGH/HomeLede #44: IPV6 设置教程

greethny, 2020-09-11:
请问是否有IPV6的详细设置教程，我是江苏移动宽带，ipv6-test.com 测试没有IPV6地址，但是路由器明明分配到了公网ipv6 ，电脑也有ipv6 就是没法访问ipv6网站，我使用电脑直接拨号是可以正常访问ipv6的

sekaiacg, 2020-09-13:
可用教程如下：https://azhuge233.com/%E5%88%B7%E5%85%A5koolshare-openwrt-x64%E4%BB%A5%E5%8F%8A%E5%90%8E%E7%BB%ADipv6%E7%9A%84%E8%AE%BE%E7%BD%AE/

chen8148, 2020-09-15:
我这里只能使用光猫拨号，按照教程中的设置过之后，ipv6-test还是不能通过

> 可用教程如下：https://azhuge233.com/%E5%88%B7%E5%85%A5koolshare-openwrt-x64%E4%BB%A5%E5%8F%8A%E5%90%8E%E7%BB%ADipv6%E7%9A%84%E8%AE%BE%E7%BD%AE/

sekaiacg, 2020-09-15:
> 我这里只能使用光猫拨号，按照教程中的设置过之后，ipv6-test还是不能通过
> 
> > 可用教程如下：https://azhuge233.com/%E5%88%B7%E5%85%A5koolshare-openwrt-x64%E4%BB%A5%E5%8F%8A%E5%90%8E%E7%BB%ADipv6%E7%9A%84%E8%AE%BE%E7%BD%AE/

光猫拨号没有试过…

sekaiacg, 2020-09-15:
网络dhcp和adg都要把禁止ipv6解析关掉

xiaoqingfengATGH, 2020-11-12:
感谢分享。如有类似需求可以参照。

tylinux, 2021-02-04:
有点儿坑，DNS 相关的组件太多了，还有个没有 GUI 配置入口的。。
1. 网络-DHCP/DNS-高级设置-禁止解析 IPV6 去掉
2. ADGuard Home-设置-DNS设置-禁止 IPV6 去掉
3. DNS-SmartDNS-勾选 IPV6 服务器 && ** 自定义设置-注释掉最后一行的 force-AAAA-SOA yes **
4. SSH 到 OpenWRT中，`vi /etc/dnscrypt-proxy2/dnscrypt-proxy.toml`, 把 `ipv6_servers`改为 true，`block_ipv6` 改为false，我顺便把 `listen_addresses` 加上了 ::1，可能非必要，懒得测试了

KyonCN, 2021-02-08:
**架构：**
主路由 ikuai 旁路由 homelede
主路由负责 dhcp 与 dhcpv6
**配置**
主路由v6 dhcp默认分配  
主路由v4 dhcp网关 dns 全都设置到旁路由
**测试**
国内v6全部可以访问 
墙外网站全部通过v4 走代理访问 
墙外未封杀v6地址段可以建立tcp链接 用来ssh
**结论**
想要全程走v6 如楼上所说 太多东西要折腾了 性价比不高 建议放弃

alexcdever, 2022-06-23:
> 有点儿坑，DNS 相关的组件太多了，还有个没有 GUI 配置入口的。。
>
> 1. 网络-DHCP/DNS-高级设置-禁止解析 IPV6 去掉
> 2. ADGuard Home-设置-DNS设置-禁止 IPV6 去掉
> 3. DNS-SmartDNS-勾选 IPV6 服务器 && **自定义设置-注释掉最后一行的 force-AAAA-SOA yes**
> 4. SSH 到 OpenWRT中，`vi [...]

反馈一下，`listen_addresses` 这个我没设置，也能有ipv6。谢谢大佬🙏