xiaoqingfengATGH/HomeLede #52: 国外无法访问

ryanhon, 2020-09-24:
## 1.passwall配置好了无法访问国外 国内都正常，用clash试了下还是一样，机场确认没问题（有问题我也来不了github了）

## 2. 详细叙述
在passwall里面的连接状态百度谷歌gihthub都能ping通但是就是无法访问，在通过教程里面的学习的ssh上路由器用
nslookup www.google.com 127.0.0.1#7053
nslookup www.google.com 127.0.0.1#54
nslookup www.google.com 127.0.0.1#53
都有返回地址，SSH内ping www.google.com也能通
但是换到电脑上就无法访问，cmd下使用nslookup www.google.com也有返回的ip，但是ping www.google.com就不通，反复确认了电脑的host文件及刷新DNS缓存都无效，很神器的问题，怀疑是ipv6惹得祸，连上光猫也改成只有ipv4的，软路由的wan6、wan和lan里面的ipv6也都关了，电脑的ipv6也关了，还是同样的问题，后来看了下homelede的dns架构图，发现我固件内的smartdns第二组也就是7053的这一组是没有开启的，但是我如果勾选启用第二DNS服务器，smartdns就无法启动了，现在不知道到底是哪出了问题，想请大佬帮忙看看能不能解决。非常喜欢大佬的固件！支持一波！

xiaoqingfengATGH, 2020-09-25:
大陆白名单模式吗？

KyonCN, 2020-09-29:
路由刷完默认应该就是没问题的 要不你试试用ikuai拨号 homelede挂旁路 感觉是passwall配置出了问题
这个固件出墙倒是没问题 只是国内网站有时候会servfail 好多个版本了 时好时坏 关闭ipv6也一样

ryanhon, 2020-10-11:
已经解决了，今天又刷了10.1，刚刷完还是一样的问题，我就在想会不会是刷系统的原因，然后就重置了一次系统，结果神奇的好了！

closed 2020-10-11 by ryanhon