xiaoqingfengATGH/HomeLede #56: 关于内网通过访问路由器域名来访问内网资源失败的问题

Raven-L, 2020-10-12:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述
  坏境：我在校园网部署了小型NAS，使用HomeLEDE作为软路由（按照说明修改了默认的LAN-IP），并使用了DDNS给域名配置了软路由的IP，另接入一个配置成AP的刷了Pandavan的MI-R3G路由器来提供WIFI。
我再两个地方遇到了改问题：
情景1：我在该服务器上搭建了NextCloud和onlyoffice套件，在连接WIFI或有线（即内网）下访问NextCloud中的onlyoffice时，会出现无法连接。而在校园网（相对于路由器是外网）中访问却可以正确进行。
情景2：我将打印机连接在了MI-R3G路由器上，并在防火墙中配置好WAN口和LAN口的端口转发后，只有WAN口的转发起作用（两个配置只有源区域的不同
[image]
）。同样是连接校园网的情况下能正常打印，在内网情况下无法打印。

本人也尝试过直接在防火墙-自定义规则中添加规则来尝试解决这个问题，但不知道是我配置没正确还是固件本身的问题，也没有成功。
另外，本人之前使用的koolshare的LEDE固件没有出现这样的问题。

### 此外，还有两个小问题
1. 固件没有自带打印机驱动，我尝试安装使用p910nd，发现打印机完全没有反应，所以把它接在了MI-R3G路由器上，然后产生了情景2的问题。我本人在HomeLEDE上安装了CUPS来解决这个问题，[解决方案](https://blog.csdn.net/Reven_L/article/details/109003476)。
2. 在使用HomeClash时，按照您的[教程](https://www.cnblogs.com/zlAurora/p/12984254.html)，没有发现HomeClash的第三方规则集
[image]

### (2) 路由器型号和固件版本
--
A：
路由器型号：EXSI下的X86软路由
固件版本：HomeLede v2020.09.05 based on OpenWrt R20.8.27 / LuCI Master (git-20.227.55359-a3340a9)
### (3) 详细日志
A：未看到有相关记录

xiaoqingfengATGH, 2020-10-13:
端口转发WAN->LAN，试试用HomeRedirect，可以解决吗？

Raven-L, 2020-10-13:
> 端口转发WAN->LAN，试试用HomeRedirect，可以解决吗？

试用了一下，可以解决问题。但HomeRedirect多个转发能力好像不太行，有几个会不运行，看了日志报错。而且这几个是随机的，会随机不运行几个。测试了一下转发三个才好像比较稳定。
[image]
[image]

xiaoqingfengATGH, 2020-10-21:
删除不运行的那几个，重新添加一次规则，能否解决？

Se7enMuting, 2020-10-27:
我也刚上手这个固件（10.1），开始也碰到同样的问题，外网可以转发访问，内网不行。。。然后电报群里的大神帮忙解决了：网络-DHCP/DNS-常规设置-“重绑定保护”的勾去掉，就好了。
然后HomeRedirect我也发现有的会不运行，基本就是同一条“目标转发地址”只有一条会生效，但是这里有不能设置这种5000-6000这样的多端口。然后我把HomeRedirect关了，在防火墙里直接设置，死活都是无效。。。。
以前一直用koolshare lede都是直接在防火墙里设置的，都没有问题。。。。

Raven-L, 2020-10-28:
> 我也刚上手这个固件（10.1），开始也碰到同样的问题，外网可以转发访问，内网不行。。。然后电报群里的大神帮忙解决了：网络-DHCP/DNS-常规设置-“重绑定保护”的勾去掉，就好了。
> 然后HomeRedirect我也发现有的会不运行，基本就是同一条“目标转发地址”只有一条会生效，但是这里又不能设置像5000-6000这样的多端口。然后我把HomeRedirect关了，在防火墙里直接设置，死活都是无效。。。。
> 以前一直用koolshare lede都是直接在防火墙里设置的，都没有问题。。。。

重绑定保护我一直都是关闭的，HomeRedirect我也没有设置多端口。HomeRedirect我感觉很看运气，一直重复应用，偶尔就能全部运行起来。koolshare lede确实没这样的问题，但感觉它可玩性已经不高了。

Se7enMuting, 2020-10-28:
> 重绑定保护我一直都是关闭的，HomeRedirect我也没有设置多端口。HomeRedirect我感觉很看运气，一直重复应用，偶尔就能全部运行起来。koolshare lede确实没这样的问题，但感觉它可玩性已经不高了。

HomeRedirect确实像你说的那样，要一直不停试。。。。问下你是在虚拟机里的吗？我是在PVE5.4理跑的~~

Raven-L, 2020-10-29:
> > > 我也刚上手这个固件（10.1），开始也碰到同样的问题，外网可以转发访问，内网不行。。。然后电报群里的大神帮忙解决了：网络-DHCP/DNS-常规设置-“重绑定保护”的勾去掉，就好了。
> > > 然后HomeRedirect我也发现有的会不运行，基本就是同一条“目标转发地址”只有一条会生效，但是这里又不能设置像5000-6000这样的多端口。然后我把HomeRedirect关了，在防火墙里直接设置，死活都是无效。。。。
> > > 以前一直用koolshare lede都是直接在防火墙里设置的，都没有问题。。。。
> > 
> > 
> > 重绑定保护我一直都是关闭的，HomeRedirect我也没有设置多端口。HomeRedirect我感觉很看运气，一直重复应用，偶尔就能全部运行起来。koolshare lede确实没这样的问题，但感觉它可玩性已经不高了。
> 
> HomeRedirect确实像你说的那样，要一直不停试。。。。问下你是在虚拟机里的吗？我是在PVE5.4里跑的~~

我是在ESXI6.7里面跑的，其实我需求也不高，也只有一个内网客户端要访问内网服务。因为我发现好像多经过一层，防火墙里的内网转发就可以生效。我是有个无线AP的，通过WIFI访问内网的转发是没问题的，但如果直接接到虚拟机本体，就会有这样的问题。

BigDog66, 2020-11-16:
> 我是在ESXI6.7里面跑的，其实我需求也不高，也只有一个内网客户端要访问内网服务。因为我发现好像多经过一层，防火墙里的内网转发就可以生效。我是有个无线AP的，通过WIFI访问内网的转发是没问题的，但如果直接接到虚拟机本体，就会有这样的问题。

这个问题楼主刚发布固件以来就存在，外网访问内网DDNS域名一直正常  内网访问内网DDNS域名不通 但如果有2个lan口 一个lan口接交换机 一个lan口接无线路由器 那么通过无线路由器接入访问域名又是正常的。。koolshare里和开发者说过很多次。如今已经快过去一年，问题还是存在。所以我一直用的是koolshare OpenWrt。今天发现koolshare楼主的贴子不见了 所以来GitHub瞧瞧  发现这个问题依然存在

tylinux, 2021-02-04:
端口转发不生效应该是 Flow Offload/SFE 加速导致，可以试试关掉加速。另，HomeRedirect 多配置起不来的，可以试试关掉总开关，应用后再打开。