xiaoqingfengATGH/HomeLede #57: dnscrypt-proxy被https-dns-proxy代替了

sekaiacg, 2020-10-14:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ x] 没有类似的issue

## 2. 详细叙述
有点疑惑，希望作者解答
### (1) 具体问题

A：homelede更新了多次dns逻辑：
1.smartdns的第二服务被dnscrypt-proxy2代替
2.10.1版本dnscrypt-proxy2被https-dns-proxy代替

smartdns与dnscrypt-proxy2都支持doh，并且有缓存功能，一直疑惑为什么不再使用，就研究了下
关于第二项，有可能是bug导致的，因为dnscrypt-proxy2启动顺序是16，https-dns-proxy是84
dnscrypt-proxy2在启动过中等待超时被https-dns-proxy后占端口7053
如果https-dns-proxy没有使用，建议固件中默认禁用

liuyuan29, 2020-10-14:
1smartdns的第二服务为什么不再使用呢？感觉速度会快点啊！点passwall的测试都是绿色的

sekaiacg, 2020-10-18:
https-dns-proxy在pw中默认启用，他和dnscrypt2冲突

xiaoqingfengATGH, 2020-10-22:
> @xiaoqingfengATGH
> https-dns-proxy在pw中默认启用，他和dnscrypt2冲突，建议not set

应该是psw最近版本刚加入特性，正在确认。

xiaoqingfengATGH, 2020-10-23:
psw内置的https-dns-proxy并不会自动启动，即使启动，端口7912，和固件默认dns端口并没有冲突。

上面提到的冲突，是怎么个现象。

sekaiacg, 2020-10-23:
> psw内置的https-dns-proxy并不会自动启动，即使启动，端口7912，和固件默认dns端口并没有冲突。
> 
> 上面提到的冲突，是怎么个现象。

端口是7053，相关提交
xiaoqingfengATGH update default bootstrap dns
Latest commit 96aa790 on 26 Apr
https://github.com/xiaoqingfengATGH/feeds-xiaoqingfeng/commit/96aa790b6abc823d9e1361f5f5a58e30d6a11ca6
系统-启动项 默认是已启用状态
我已经手动no set了，所以没有相关截图，好像原因不明了

xiaoqingfengATGH, 2020-10-25:
问题原因已经找到。

Passwall支持使用https-dns-proxy作为dns解析备选方案，因此会自动安装https-dns-proxy。
而固件源中自带的https-dns-proxy被安装后会使用默认自带配置启动，并且占用7053端口。
占用后会影响固件自带的dnscrpyt-proxy的启动，导致海外DNS解析失败，进而无法访问海外网站。

目前已经修正。

修正方法如下：

固件源中自带的https-dns-proxy不再安装自动启动脚本，同时做了一个升级，版本升级到了2020-09-25
重新编译固件，即可。

已经有固件，遇到问题，
请执行rm -rf /etc/init.d/https-dns-proxy，随后重启固件。

closed 2020-10-25 by xiaoqingfengATGH