xiaoqingfengATGH/HomeLede #60: 断电后的恢复

Wubowen1984, 2020-10-21:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [X] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：用QNAP的NAS上的虚拟机运行，目前使用v2020.10.01 测试版，NAS断电，来电后NAS启动后，必须要重启Homelede一次才能连接PPPOE，在接口中停止WAN后在连接，能看到WAN口IP变化，但不能访问网站，重启后可正常访问

### (2) 路由器型号和固件版本

A：NAS是QNAP的TS-453Bmini，固件版本v2020.10.01 测试版

### (3) 详细日志

A：

KyonCN, 2020-10-23:
我在PVE下使用没有问题 1001感觉是半年来最稳定的版本了 再也没有出现过dns serfail的问题

Wubowen1984, 2020-10-24:
要注意什么配置么？

xiaoqingfengATGH, 2020-10-29:
pppoe重启时会自动连接。

断开后也会自动重连。

目前我遇到的pppoe不连接的情况大多和多拨有关。

多拨时，利用了isp漏洞，同时发起多个连接，并行确认。成功后，如果某个连接断掉，会自动重播，但此时因为已经超过isp连接上限会出现无法连接情况。结果就是发现多拨后，似乎无法重播掉线的连接。
实际上，多拨对拨号时间要求比较高，得所有都断开后，再一起重播。

isp对连接频率有限制。已经拨号成功，如果重启路由，再次拨号，可能发现短时间内无法拨号现象。
因为isp端认为你仍旧在线，会拒绝拨号。需要等待几分钟后，再次重播。

binggz, 2020-11-01:
### 请解决docker的一个容器兼容问题
需要在编译时解决，例如： [https://www.right.com.cn/forum/thread-4052047-1-1.html](url)
谢谢！
```
root@OpenWrt:~# ip link set br-lan promisc on
root@OpenWrt:~# docker run --rm --privileged multiarch/qemu-user-static --reset -p yes
Unable to find image 'multiarch/qemu-user-static:latest' locally
latest: Pulling from multiarch/qemu-user-static
9758c28807f2: Pull complete
1541dc593e72: Pull complete
ed8a5179ae11: Pull complete
1ec39da9c97d: Pull complete
ebb26d08461b: Pull complete
Digest: sha256:a9944443ade713bc12ca0579c0e15a090ab43aa7a0d7b4c6883d1fbd44f6902f
Status: Downloaded newer image for multiarch/qemu-user-static:latest
No binfmt support in the kernel.
  Try: '/sbin/modprobe binfmt_misc' from the host

```

sekaiacg, 2020-11-10:
> ### 请解决docker的一个容器兼容问题
> 需要在编译时解决，例如： [https://www.right.com.cn/forum/thread-4052047-1-1.html](url)
> 谢谢！
>
> ```
> root@OpenWrt:~# ip link set br-lan promisc on
> root@OpenWrt:~# docker run --rm --privileged [...]

@binggz @xiaoqingfengATGH 
尝试过，可能出现问题
1.make kernel_menuconfig会覆盖内核设置
2.make -j8 V=s 编译homelede又会覆盖1中的设置
3.没有仔细研究lede的内核配置与make kernel_menuconfig有何区别，但是分别生成的2个.config差异巨大

**以下方法从步骤8.make -j8 V=s开始，编译教程见homeled主页**
**解决方法一：修改已有的内核配置**
1.执行make -j8 V=s生成内核配置后ctrl+c终止编译
2.修改build_dir/target-x86_64_musl/linux-x86_64/linux-5.4.72/.config
_**配置**_
`# CONFIG_BINFMT_MISC is not set`
_**改为**_
`CONFIG_BINFMT_MISC=y`
3.执行make -j8 V=s重新编译

**解决方法二：新增配置选项(作者添加)**

1.在config/Config-kernel.in添加配置选项
```
config KERNEL_BINFMT_MISC
	bool "Enable support for binfmt"
	default y
	help
	  binfmt_misc (Miscellaneous Binary Format) is a capability of the
	  Linux kernel which allows arbitrary executable file formats to be
	  recognized and passed to certain user space applications, such as
	  emulators and virtual machines.
```
2.执行命令make menuconfig，出现界面后什么也不做ctrl+c退出
3.删除已编译的内核缓存：build_dir/target-x86_64_musl/linux-x86_64/linux-5.4.72
4.开始重新编译：make -j8 V=s