xiaoqingfengATGH/HomeLede #63: 请教：直连列表里面的网址还是通过代理

TyrielWW, 2020-11-05:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
[ * ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：PassWall里面，直连列表里面的网址还是通过代理
在直连列表中添加了网址，没有加http://www前缀。通过网站查看访问ip，发现走的是代理。
模式设置的是“中国列表以外”。

改过IP地址，把192.168.1.1改成192.168.1.2，DHCP关了。其他设置没动过，不知道这个有没有影响。

### (2) 路由器型号和固件版本

A：软路由，PVE建的虚拟机，2核1G，固件：2020.10.01；2020.07.18。

### (3) 详细日志

A：2020-11-05 09:46:59: 执行启动延时 1 秒后再启动!
2020-11-05 09:47:00: 分析 Socks 服务的节点配置...
2020-11-05 09:47:00: TCP_1节点：gcore.aiji.ru，节点：www.abc.com:443，监听端口：1041
2020-11-05 09:47:00: UDP_1节点：gcore.aiji.ru，节点：www.abc.com:443，监听端口：1051
2020-11-05 09:47:00: 过滤服务配置：准备接管域名解析[0]...
2020-11-05 09:47:00:   | - (homelede -> chinadns-ng) 只支持2~4级的域名过滤...
2020-11-05 09:47:00:   | - (homelede -> chinadns-ng) 上游 DNS (127.0.0.1#7053) 有一定概率会比 DNS (127.0.0.1#6053) 先返回的话(比如 DNS 的本地查询缓存)，启用 '公平模式' 可以优先接受代理 DNS 的中国IP段内(chnroute/chnroute6)的应答...
2020-11-05 09:47:00:   + 过滤服务：HomeLede->ChinaDNS-NG(:7913-f) + 代理：中国域名列表：127.0.0.1#6053，防火墙域名列表：127.0.0.1#7053
2020-11-05 09:47:01:   - [0]节点列表中的域名(vpsiplist)：119.29.29.29
2020-11-05 09:47:01:   - [0]域名白名单(whitelist)：默认
2020-11-05 09:47:01:   - [1]代理域名表(blacklist)：默认
2020-11-05 09:47:01:   - [0]防火墙域名表(gfwlist)：默认
2020-11-05 09:47:01:   - [0]以上所列以外及默认(homelede -> chinadns-ng)：127.0.0.1#7913
2020-11-05 09:47:01: 开始加载防火墙规则...
2020-11-05 09:47:01: 本机网段互访直连：192.168.1.2/24
2020-11-05 09:47:01: 过滤所有节点直接 IP 地址完成[0]
2020-11-05 09:47:01: 加载路由器自身 TCP 代理...
2020-11-05 09:47:01: 使用链表 PREROUTING 排列索引2[0]
2020-11-05 09:47:01: 分析 Socks 服务所使用节点...
2020-11-05 09:47:01: 分析 TCP 代理自动切换1...
2020-11-05 09:47:06:   - [2]TCP直连代理使用链2，节点（v2ray）：www.abc.com:443
2020-11-05 09:47:06: 分析 UDP 代理自动切换1...
2020-11-05 09:47:11:   - [2]UDP直连代理使用链2，节点（v2ray）：www.abc.com:443
2020-11-05 09:47:11: 加载路由器自身 UDP 代理...
2020-11-05 09:47:11:   - [0]不代理 UDP 端口：80,443
2020-11-05 09:47:11: TCP默认代理：使用TCP节点1 [中国列表以外](REDIRECT:1041)代理所有端口
2020-11-05 09:47:11: UDP默认代理：使用UDP节点1 [中国列表以外](TPROXY:1051)代理除disable外的所有端口
2020-11-05 09:47:11: 防火墙规则加载完成！
2020-11-05 09:47:11: 重启 dnsmasq 服务[0]
2020-11-05 09:47:11: 配置定时任务：自动更新规则。
2020-11-05 09:47:11: 运行完成！

xiaoqingfengATGH, 2020-11-05:
DNS页面，清空一下ipset试试？

TyrielWW, 2020-11-05:
试了清空ipset，不管用

qq66711607, 2021-05-20:
今天遇到同样的问题

dreamcreated, 2022-05-23:
遇到同样的问题

sleepystrigiformes, 2022-09-28:
[image]
高级设置里我这么改完就好了。

Amanoshoyichi, 2022-12-11:
是的，我今天也发现了这个问题。

chenzhiguo, 2023-12-30:
附议

woeoio, 2024-04-06:
坐等最终方案