xiaoqingfengATGH/HomeLede #64: 关于docker提示binfmt_misc的解决方法

sekaiacg, 2020-11-10:
**以下方法从编译固件步骤8.make -j8 V=s开始，编译教程见homeled主页**
`${kernel_version}`为issue发布前源码内核版本：**5.4.72**
`解决方法一：修改已有的内核配置`
1.执行make -j8 V=s生成内核配置后ctrl+c终止编译
2.修改build_dir/target-x86_64_musl/linux-x86_64/${kernel_version}/.config
**_配置_**
`# CONFIG_BINFMT_MISC is not set`
**_改为_**
`CONFIG_BINFMT_MISC=y`
3.执行make -j8 V=s重新编译

**解决方法二：新增配置选项(作者添加@xiaoqingfengATGH)**
1.在config/Config-kernel.in添加配置选项
```
config KERNEL_BINFMT_MISC
	bool "Enable support for binfmt"
	default y
	help
	  binfmt_misc (Miscellaneous Binary Format) is a capability of the
	  Linux kernel which allows arbitrary executable file formats to be
	  recognized and passed to certain user space applications, such as
	  emulators and virtual machines.
```
2.执行命令make menuconfig，出现界面后什么也不做ctrl+c退出
3.删除已编译的内核缓存：build_dir/target-x86_64_musl/linux-x86_64/${kernel_version}
4.开始重新编译：make -j8 V=s

sekaiacg, 2020-11-10:
@binggz

xiaoqingfengATGH, 2020-11-11:
感谢分享。有类似需求的朋友可按照上述描述调整编译。