xiaoqingfengATGH/HomeLede #86: VPN 无法连接

g199209, 2021-02-11:
客户端是 MAC 电脑，连接始终失败，手动运行 strongswan 得到以下错误 Log：

```
07[IKE] 10.42.100.7 is initiating a Main Mode IKE_SA
07[CFG] selected proposal: IKE:AES_CBC_128/HMAC_SHA1_96/PRF_HMAC_SHA1/MODP_1024
07[ENC] generating ID_PROT response 0 [ SA V V V V ]
07[NET] sending packet: from 10.42.240.30[500] to 10.42.100.7[500] (160 bytes)
13[NET] received packet: from 10.42.100.7[500] to 10.42.240.30[500] (228 bytes)
13[ENC] parsed ID_PROT request 0 [ KE No NAT-D NAT-D ]
13[ENC] generating ID_PROT response 0 [ KE No NAT-D NAT-D ]
13[NET] sending packet: from 10.42.240.30[500] to 10.42.100.7[500] (244 bytes)
09[NET] received packet: from 10.42.100.7[500] to 10.42.240.30[500] (108 bytes)
09[ENC] parsed ID_PROT request 0 [ ID HASH N(INITIAL_CONTACT) ]
09[CFG] looking for pre-shared key peer configs matching 10.42.240.30...10.42.100.7[10.42.100.7]

09[IKE] found 1 matching config, but none allows pre-shared key authentication using Main Mode

09[ENC] generating INFORMATIONAL_V1 request 1020812564 [ HASH N(AUTH_FAILED) ]
09[NET] sending packet: from 10.42.240.30[500] to 10.42.100.7[500] (92 bytes)
```

错误原因应该是  `09[IKE] found 1 matching config, but none allows pre-shared key authentication using Main Mode`

手动修改 `ipsec.conf` 文件：

```
conn xauth_psk
        keyexchange=ikev1
        ike=aes128-sha1-modp2048,aes128-sha1-modp1024,3des-sha1-modp1024,3des-sha1-modp1536
        esp=aes128-sha1,3des-sha1
        left=%defaultroute
        leftauth=psk
        leftsubnet=0.0.0.0/0
        right=%any
        rightauth=psk
         # rightauth2=xauth
        rightsourceip=10.42.17.100/24
        rightdns=10.42.0.1
        auto=add
```

把 `rightauth2=xauth` 去掉，此时上述错误没有了，然而又有其他错误了：

```
11[IKE] 10.42.100.7 is initiating a Main Mode IKE_SA
11[CFG] selected proposal: IKE:AES_CBC_128/HMAC_SHA1_96/PRF_HMAC_SHA1/MODP_1024
11[ENC] generating ID_PROT response 0 [ SA V V V V ]
11[NET] sending packet: from 10.42.240.30[500] to 10.42.100.7[500] (160 bytes)
10[NET] received packet: from 10.42.100.7[500] to 10.42.240.30[500] (228 bytes)
10[ENC] parsed ID_PROT request 0 [ KE No NAT-D NAT-D ]
10[ENC] generating ID_PROT response 0 [ KE No NAT-D NAT-D ]
10[NET] sending packet: from 10.42.240.30[500] to 10.42.100.7[500] (244 bytes)
07[NET] received packet: from 10.42.100.7[500] to 10.42.240.30[500] (108 bytes)
07[ENC] parsed ID_PROT request 0 [ ID HASH N(INITIAL_CONTACT) ]
07[CFG] looking for pre-shared key peer configs matching 10.42.240.30...10.42.100.7[10.42.100.7]
07[CFG] selected peer config "xauth_psk"
07[IKE] IKE_SA xauth_psk[1] established between 10.42.240.30[10.42.240.30]...10.42.100.7[10.42.100.7]
07[IKE] scheduling reauthentication in 10168s
07[IKE] maximum IKE_SA lifetime 10708s
07[ENC] generating ID_PROT response 0 [ ID HASH ]
07[NET] sending packet: from 10.42.240.30[500] to 10.42.100.7[500] (76 bytes)
14[NET] received packet: from 10.42.100.7[500] to 10.42.240.30[500] (284 bytes)
14[ENC] parsed QUICK_MODE request 2268403166 [ HASH SA No ID ID ]

14[IKE] no matching CHILD_SA config found for 10.42.100.7/32[udp/50230] === 10.42.240.30/32[udp/l2f]

14[ENC] generating INFORMATIONAL_V1 request 1904265780 [ HASH N(INVAL_ID) ]
14[NET] sending packet: from 10.42.240.30[500] to 10.42.100.7[500] (76 bytes)
09[NET] received packet: from 10.42.100.7[500] to 10.42.240.30[500] (284 bytes)
09[IKE] received retransmit of request with ID 2268403166, but no response to retransmit
08[NET] received packet: from 10.42.100.7[500] to 10.42.240.30[500] (284 bytes)
08[IKE] received retransmit of request with ID 2268403166, but no response to retransmit
12[NET] received packet: from 10.42.100.7[500] to 10.42.240.30[500] (284 bytes)
12[IKE] received retransmit of request with ID 2268403166, but no response to retransmit
15[NET] received packet: from 10.42.100.7[500] to 10.42.240.30[500] (284 bytes)
15[IKE] received retransmit of request with ID 2268403166, but no response to retransmit
06[NET] received packet: from 10.42.100.7[500] to 10.42.240.30[500] (284 bytes)
06[IKE] received retransmit of request with ID 2268403166, but no response to retransmit
```

`14[IKE] no matching CHILD_SA config found for 10.42.100.7/32[udp/50230] === 10.42.240.30/32[udp/l2f]`，看样子是没有把 L2TP 用的 1701 端口正确转发？

求问下此时该如何解决？

g199209, 2021-02-11:
配置了 ikev2，认证是正常通过了，然而有其他问题，会显示 `05[KNL] received netlink error: No such file or directory (2)`，
和这个 issue 中的情况基本一样：

https://github.com/openwrt/packages/issues/14360

xiaoqingfengATGH, 2021-02-18:
strongswan其实还有其他问题，近期打算换vpn方案了，后面版本体验吧

closed 2021-02-18 by xiaoqingfengATGH