xiaoqingfengATGH/HomeLede #89: UPNP让我自闭

gnightboy, 2021-02-16:
这次斗胆安装了一下最新测试版！

直呼好家伙，SFE不见了，UPNP不见了，SSPLUS不见了。  

折腾了4个小时UPNP，一直提示“正在收集数据....”，不管怎么重启，怎么按照网上改配置，都是正在收集数据，迅雷下载只有之前的10分之1的速度了。

h0wardch3ng, 2021-02-17:
我放弃这个系统了，wiki里面也是一堆问题
1.esxi虚拟机的分区看了一下只有sda1，sda2是不存在的，wiki还提示新建扩容分区的sda3
2.我不用虚拟机方式安装，直接写盘，无论是img.gz还是img都无法正常写盘

gnightboy, 2021-02-17:
> 我放弃这个系统了，wiki里面也是一堆问题
> 1.esxi虚拟机的分区看了一下只有sda1，sda2是不存在的，wiki还提示新建扩容分区的sda3
> 2.我不用虚拟机方式安装，直接写盘，无论是img.gz还是img都无法正常写盘

以前的版本我用着很顺， 有点追新，新内核版本有点点小问题，不过还能接受。

KyonCN, 2021-02-17:
如果你的cpu不是特别拉胯 sfe不用开 千兆过墙服务下也就5%cpu占用 
开了反而问题挺多的

xiaoqingfengATGH, 2021-02-18:
测试版就是让大家发现问题，有问题积极反馈。1 Ssr plus历来没有集成，都是软件包提供。2 sfe因为很大几率导致死机等问题，且性能提升有限，已经不再默认集成，但软件包提供。3 upnp没有集成是失误。可以手动设置转发解决。 @gnightboy

xiaoqingfengATGH, 2021-02-18:
> 我放弃这个系统了，wiki里面也是一堆问题
> 1.esxi虚拟机的分区看了一下只有sda1，sda2是不存在的，wiki还提示新建扩容分区的sda3
> 2.我不用虚拟机方式安装，直接写盘，无论是img.gz还是img都无法正常写盘

wiki是领路的，有问题要变通处理，建议补充下linux关于磁盘的基本知识。也可以用0905版本，可能完全匹配教程。

gnightboy, 2021-02-18:
> 测试版就是让大家发现问题，有问题积极反馈。1 Ssr plus历来没有集成，都是软件包提供。2 sfe因为很大几率导致死机等问题，且性能提升有限，已经不再默认集成，但软件包提供。3 upnp没有集成是失误。可以手动设置转发解决。 @gnightboy

@xiaoqingfengATGH 谢谢，已经重刷稳定版了，等下一个测试版。

gnightboy, 2021-02-18:
> 如果你的cpu不是特别拉胯 sfe不用开 千兆过墙服务下也就多5%cpu占用
> 开了反而问题挺多的

@KyonCN   BBR对过墙影响大吗？   其实我觉得难受的是UPNP，迅雷下载速度变慢了。

KyonCN, 2021-02-18:
> > 如果你的cpu不是特别拉胯 sfe不用开 千兆过墙服务下也就多5%cpu占用
> > 开了反而问题挺多的
> 
> @KyonCN BBR对过墙影响大吗？ 其实我觉得难受的是UPNP，迅雷下载速度变慢了。

在线路面前一切都是浮云  路由器端的BBR基本没啥提升 不会像服务器那么明显

PlayerYao, 2021-03-11:
> > 测试版就是让大家发现问题，有问题积极反馈。1 Ssr plus历来没有集成，都是软件包提供。2 sfe因为很大几率导致死机等问题，且性能提升有限，已经不再默认集成，但软件包提供。3 upnp没有集成是失误。可以手动设置转发解决。 @gnightboy
> 
> @xiaoqingfengATGH 谢谢，已经重刷稳定版了，等下一个测试版。

你刷的哪个稳定版 我的也没有了Upnp很难受