xiaoqingfengATGH/HomeLede #92: dnsmasq 处理 DNS 解析时错误的过滤了某些地址

g199209, 2021-02-21:
为方便访问内网设备，我在公共 DNS 上配置了某些域名解析到内网地址，然而发现若 DNS 解析流程通过 dnsmasq 会有异常。

内网网段：`10.42.0.1/16`

dig 测试结果如下：

```
root@OpenWrt:/etc/chinadns-ng# dig gogs.gaomf.store @127.0.0.1 -p 5053

; <<>> DiG 9.16.4 <<>> gogs.gaomf.store @127.0.0.1 -p 5053
;; global options: +cmd
;; Got answer:
;; ->>HEADER<<- opcode: QUERY, status: NOERROR, id: 50376
;; flags: qr rd ra; QUERY: 1, ANSWER: 1, AUTHORITY: 0, ADDITIONAL: 1

;; OPT PSEUDOSECTION:
; EDNS: version: 0, flags:; udp: 512
;; QUESTION SECTION:
;gogs.gaomf.store.		IN	A

;; ANSWER SECTION:
gogs.gaomf.store.	1652	IN	A	10.42.2.21

;; Query time: 0 msec
;; SERVER: 127.0.0.1#5053(127.0.0.1)
;; WHEN: Sun Feb 21 18:13:27 CST 2021
;; MSG SIZE  rcvd: 61
```

```
root@OpenWrt:/etc/chinadns-ng# dig gogs.gaomf.store @127.0.0.1 -p 54

; <<>> DiG 9.16.4 <<>> gogs.gaomf.store @127.0.0.1 -p 54
;; global options: +cmd
;; Got answer:
;; ->>HEADER<<- opcode: QUERY, status: NOERROR, id: 136
;; flags: qr rd ra; QUERY: 1, ANSWER: 0, AUTHORITY: 0, ADDITIONAL: 1

;; OPT PSEUDOSECTION:
; EDNS: version: 0, flags:; udp: 512
;; QUESTION SECTION:
;gogs.gaomf.store.		IN	A

;; Query time: 4 msec
;; SERVER: 127.0.0.1#54(127.0.0.1)
;; WHEN: Sun Feb 21 18:14:32 CST 2021
;; MSG SIZE  rcvd: 45
```

可以看到不知为何 dnsmasq 把 gogs.gaomf.store 的查询结果过滤掉了。

目前使用的 walk around 方案是在 AdGuard 中把上游服务器配置为 `127.0.0.1:5053` 而不是 `127.0.0.1:54`.

不知此问题的原因是什么，有没有什么更好些的解决方案？

huxim, 2021-06-09:
我也发现了，54端口过滤了很多域名
https://twitter.com/huxim/status/1402188845286756353

g199209, 2023-04-02:
> 目前方案中，dnsmasq已经去除了所有内置列表，并不会过滤或者自行解析域名。其行为交给分流软件，如passwall，clash等控制。

使用 v2023.02.05 测试，此问题依然未解决。而且 `chinadns-ng` 的监听端口换了一下，目前要把 AdGuard 中的上游服务器设置为 `127.0.0.1:15354` 才可以正常使用。