xiaoqingfengATGH/HomeLede #102: 在“passwall代理列表”中增加并保存和应用了一条网址后，整个直连代理名单全部清空了

waynesparrow123, 2021-03-26:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：在“passwall代理列表”中增加并保存和应用了一条网址后，整个直连代理名单全部清空了

### (2) 路由器型号和固件版本

A：
主机名 | OpenWrt
主机型号 | Intel(R) Celeron(R) CPU 3965U @ 2.20GHz : 2 Core 2 Thread (CpuMark : 33672.631867 Scores)
CPU 信息 | 2200.034 MHz +39.0°C (high = +100.0°C, crit = +100.0°C)
固件版本 | HomeLede v2021.03.20 based on OpenWrt R21.2.1 / LuCI Master (git-21.037.51918-08b66fb)
内核版本 | 5.4.105
本地时间 | Fri Mar 26 23:27:37 2021

### (3) 详细日志

A：
Fri Mar 26 18:00:00 2021 cron.err crond[6765]: USER root pid 22328 cmd [ -x /etc/init.d/smartdns ] && /etc/init.d/smartdns restart > /dev/null 2>&1 &
Fri Mar 26 18:53:10 2021 daemon.err nlbwmon[4002]: Netlink receive failure: Out of memory
Fri Mar 26 18:53:10 2021 daemon.err nlbwmon[4002]: Unable to dump conntrack: No buffer space available
Fri Mar 26 19:00:12 2021 user.info mwan3track[3148]: Check (ping) failed for target "114.114.114.114" on interface wan (eth0). Current score: 11
Fri Mar 26 19:00:14 2021 user.info mwan3track[3148]: Check (ping) failed for target "www.baidu.com" on interface wan (eth0). Current score: 11
Fri Mar 26 19:00:14 2021 user.notice mwan3track[3148]: Interface wan (eth0) is disconnecting
Fri Mar 26 19:00:21 2021 user.info mwan3track[3148]: Check (ping) failed for target "119.29.29.29" on interface wan (eth0). Current score: 10
Fri Mar 26 19:00:23 2021 user.info mwan3track[3148]: Check (ping) failed for target "www.baidu.com" on interface wan (eth0). Current score: 10
Fri Mar 26 19:00:28 2021 user.notice mwan3track[3148]: Interface wan (eth0) is connecting
Fri Mar 26 19:00:28 2021 user.info mwan3track[3148]: Lost 4 ping(s) on interface wan (eth0). Current score: 9
Fri Mar 26 19:08:22 2021 user.info mwan3track[3148]: Check (ping) failed for target "119.29.29.29" on interface wan (eth0). Current score: 11
Fri Mar 26 20:16:21 2021 user.info mwan3track[3148]: Check (ping) failed for target "119.29.29.29" on interface wan (eth0). Current score: 11
Fri Mar 26 21:03:14 2021 daemon.err dnscrypt-proxy[2441]: [2021-03-26 13:03:14] [NOTICE] [moulticast-sg-ipv4] TIMEOUT
Fri Mar 26 21:03:31 2021 daemon.err dnscrypt-proxy[2441]: [2021-03-26 13:03:31] [NOTICE] [meganerd.DNSCrypt] TIMEOUT
Fri Mar 26 21:03:36 2021 daemon.err dnscrypt-proxy[2441]: [2021-03-26 13:03:36] [NOTICE] [moulticast-sg-ipv4] TIMEOUT
Fri Mar 26 21:03:41 2021 daemon.err dnscrypt-proxy[2441]: [2021-03-26 13:03:41] [NOTICE] [meganerd.DNSCrypt] TIMEOUT
Fri Mar 26 21:03:42 2021 daemon.err dnscrypt-proxy[2441]: [2021-03-26 13:03:42] [NOTICE] Sorted latencies:
Fri Mar 26 21:03:42 2021 daemon.err dnscrypt-proxy[2441]: [2021-03-26 13:03:42] [NOTICE] -    71ms google.doh
Fri Mar 26 21:03:42 2021 daemon.err dnscrypt-proxy[2441]: [2021-03-26 13:03:42] [NOTICE] -    99ms nextdns.doh
Fri Mar 26 21:03:42 2021 daemon.err dnscrypt-proxy[2441]: [2021-03-26 13:03:42] [NOTICE] -   127ms cloudflare.doh
Fri Mar 26 21:03:42 2021 daemon.err dnscrypt-proxy[2441]: [2021-03-26 13:03:42] [NOTICE] -   216ms moulticast-sg-ipv4
Fri Mar 26 21:03:42 2021 daemon.err dnscrypt-proxy[2441]: [2021-03-26 13:03:42] [NOTICE] -   236ms meganerd.DNSCrypt
Fri Mar 26 21:03:42 2021 daemon.err dnscrypt-proxy[2441]: [2021-03-26 13:03:42] [NOTICE] Server with the lowest initial latency: google.doh (rtt: 71ms)
Fri Mar 26 21:18:29 2021 user.info mwan3track[3148]: Check (ping) failed for target "119.29.29.29" on interface wan (eth0). Current score: 11
Fri Mar 26 21:41:13 2021 daemon.err nlbwmon[4002]: Netlink receive failure: Out of memory
Fri Mar 26 21:41:13 2021 daemon.err nlbwmon[4002]: Unable to dump conntrack: No buffer space available
Fri Mar 26 22:51:48 2021 daemon.err uhttpd[3852]: cat: write error: Broken pipe
Fri Mar 26 22:51:58 2021 cron.err crond[25530]: crond (busybox 1.31.1) started, log level 5
Fri Mar 26 22:51:58 2021 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Fri Mar 26 22:51:58 2021 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Fri Mar 26 22:52:01 2021 daemon.info dnsmasq[25664]: Connected to system UBus
Fri Mar 26 22:52:03 2021 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Fri Mar 26 22:52:03 2021 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Fri Mar 26 22:52:06 2021 cron.err crond[27494]: crond (busybox 1.31.1) started, log level 5
Fri Mar 26 22:52:06 2021 daemon.info dnsmasq[27400]: Connected to system UBus
Fri Mar 26 22:52:07 2021 cron.err crond[27734]: crond (busybox 1.31.1) started, log level 5
Fri Mar 26 22:52:07 2021 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Fri Mar 26 22:52:07 2021 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Fri Mar 26 22:52:10 2021 daemon.info dnsmasq[27875]: Connected to system UBus
Fri Mar 26 22:52:12 2021 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Fri Mar 26 22:52:12 2021 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Fri Mar 26 22:52:15 2021 cron.err crond[29704]: crond (busybox 1.31.1) started, log level 5
Fri Mar 26 22:52:15 2021 daemon.info dnsmasq[29610]: Connected to system UBus
Fri Mar 26 23:01:00 2021 cron.err crond[29704]: USER root pid 30776 cmd sh /usr/share/jd-dailybonus/newapp.sh -u
Fri Mar 26 23:10:24 2021 daemon.err uhttpd[3852]: netstat: showing only processes with your user ID

yuntian123456, 2021-03-28:
给你划重点
Fri Mar 26 21:41:13 2021 daemon.err nlbwmon[4002]: Netlink receive failure: **Out of memory**
Fri Mar 26 21:41:13 2021 daemon.err nlbwmon[4002]: Unable to dump conntrack: **No buffer space available**

dfbb9216, 2022-03-08:
overlay扩容