xiaoqingfengATGH/HomeLede #104: home connect新建用户异常

alexcdever, 2021-04-17:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述
启用home connect后，状态变更为“运行中”， 与共享密钥也可以保存成功，但添加用户反而会失败，显示的错误为 `Error code 8: The specified Virtual Hub does not exist on the server.`

### (1) 具体问题

A：我尝试了关闭再启用，但是还是不行。另外我还试图卸载再安装SoftEther，不过重装后页面里见不到home connect里，请问怎么重新显示？我试过删除/tmp/luci-indexcache, 但还是没显示

### (2) 路由器型号和固件版本

A：PVE下通过LXT安装的x86软路由，固件是04-05测试版

### (3) 详细日志

A：不清楚这方面的日志该去哪里查

xiaoqingfengATGH, 2021-04-22:
你如果自行安装过softether，可能会破坏homeconnect中内置的配置。

其中有个virtualhub，名称为homelede，你可能删除了。

可以去下面下载默认配置。
https://raw.githubusercontent.com/xiaoqingfengATGH/feeds-xiaoqingfeng/master/softethervpn5/files/vpn_server.config

将下载后的文件覆盖/usr/libexec/softethervpn/vpn_server.config

重启SoftEther

alexcdever, 2021-05-26:
不好意思，前段时间一直在忙，都没留意大佬你的回复。我现在是把homelede部署在了vm里，aio还是太麻烦了，后面打算重新找个路由器刷机好了。谢谢大佬回复

closed 2021-05-26 by alexcdever

yishion819, 2021-11-24:
> 你如果自行安装过softether，可能会破坏homeconnect中内置的配置。
> 
> 其中有个virtualhub，名称为homelede，你可能删除了。
> 
> 可以去下面下载默认配置。 https://raw.githubusercontent.com/xiaoqingfengATGH/feeds-xiaoqingfeng/master/softethervpn5/files/vpn_server.config
> 
> 将下载后的文件覆盖/usr/libexec/softethervpn/vpn_server.config
> 
> 重启SoftEther

我也出现这个问题了  我重新下载这个配置后 重启了路由器 [还是提示这个错误 因为我以前是用ipsecvpn ikev2 但这个IKEV2跟NAT有冲突  我关闭了整个ipsecvpn 重新打开的connect 新建用户就出错了