xiaoqingfengATGH/HomeLede #107: homedirect设置多条时，不能正常启动

for5million, 2021-05-20:
反馈bug/问题模板

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：设置了50080到openwrt，50081到unraid的80，53389到win10的3389
全部勾选，点击保存并应用，不一定那个是生效的

有时是第1条，有时是2，3条，有时只有第3条

### (2) 路由器型号和固件版本

A：软路由 4405u 6口 211网卡

### (3) 详细日志

A：不知道从哪里查

for5million, 2021-05-20:
截图：
[image]

Mr-dbo, 2021-05-25:
那是因为你连接着这些端口，不要连接这些端口，比如，3389 ，先不要远程。然后保存，就能成功。这个应该是BUG，我已经弃用这个了，用其他端口转发工具。

xiaoqingfengATGH, 2021-06-01:
原因多次启动同一个端口转发，导致之前占用的端口没有释放。

HomeRedirect底层基于socat，配置一条转发规则，相当于开启一个在路由上的TCP SERVER（比如你转发58080），则会占用路由58080开启一个TCP LISTEN，此时如果多次应用转发规则，HomeRedirect会关闭之前的TCP Server，再重新开启。

由于TCP（Server）关闭时，会有一个1-4分钟的TIME WAIT的状态，会导致已经使用过的端口短时间内不能再次使用。此时再启动转发就会失败。

1 可以尝试配置好转发后，重启路由。
2 也可以修改路由的TCP参数，改善这个问题。
```
net.ipv4.tcp_syncookies = 1 表示开启SYN Cookies。当出现SYN等待队列溢出时，启用cookies来处理，可防范少量SYN攻击，默认为0，表示关闭；
net.ipv4.tcp_tw_reuse = 1 表示开启重用。允许将TIME-WAIT sockets重新用于新的TCP连接，默认为0，表示关闭；
net.ipv4.tcp_tw_recycle = 1 表示开启TCP连接中TIME-WAIT sockets的快速回收，默认为0，表示关闭。
net.ipv4.tcp_fin_timeout 修改系默认的 TIMEOUT 时间
```
然后执行 /sbin/sysctl -p 让参数生效.

binbinsan, 2021-06-25:
这边也同样出现了这个问题，设置端口转发，可正常使用后，通过外网打开一次转发后，再开启网页就开启不起来了。
系统页面也显示系统未运行。
之前版本记得没有此问题的。

binbinsan, 2021-06-25:
通过内部NAT环回没有此问题，再局域网内，打开外部映射端口。环回没问题，就是外网访问不行，页面只能打开一次。后边就打不开了。麻烦您有时间检查代码，感谢。
另，这边git这份代码，变异会报错，通过不了。

mzdsxgl, 2021-06-25:
打开/etc/config/homeredirect
把对应的option src_ip 改成0.0.0.0就可以了

binbinsan, 2021-06-25:
收到，感谢回复。这边测试一下。

binbinsan, 2021-06-26:
> 打开/etc/config/homeredirect
> 把的选项src_ip改成0.0.0.0就可以了

[image]
我打开这个配置文件，没有找到您说的条目，是需要手动新增么？

binbinsan, 2021-06-26:
> 打开/etc/config/homeredirect
> 把对应的option src_ip 改成0.0.0.0就可以了

刚才我测试了一下，手动新增此条目之后，
路由器后端：192.168.123.152：端口转发无法访问，但是可以访问192.168.123.1本机
刚才要打开后端访问，必须要手动设置，防火墙转发规则，但是这样，内网nat环回对于192.168.123.152：就没法用了，
对192.168.123.1依然有效
[image]

mzdsxgl, 2021-06-27:
> > 打开/etc/config/homeredirect
> > 把对应的option src_ip 改成0.0.0.0就可以了
> 
> 刚才我测试了一下，手动新增此条目之后，
> 路由器后端：192.168.123.152：端口转发无法访问，但是可以访问192.168.123.1本机
> 刚才要打开后端访问，必须要手动设置，防火墙转发规则，但是这样，内网nat环回就没法用了。
> [image]

情况还有点不一样
我本来也是多个配置无法全部生效，查看了配置以后发现

_config redirect
	option src_ip '::'
	option dest_port '3389'_

src_ip这个地方不正确，手工改成0.0.0.0之后就好了，目前转发正常。

最新613这个版本我碰到好几个问题。
比如重启之后，客户端无法正常DNS解析，再次重启又好了。
又比如经常发现无法修改配置，实质是文件系统read-only。
能力有效，无法解决这些问题，我现在退回320版本了，不知道是不是会稳定一点。

binbinsan, 2021-06-27:
> > > 打开/etc/config/homeredirect
> > > 把的选项src_ip改成0.0.0.0就可以了
> > 
> > 
> > 刚我测试了一下，手动此条目之后，
> > 还有很多：192.168.123.152：端口转发无法访问，但可以访问192.168.123.1本
> > 机要打开完整访问，必须手动设置，防火墙转发规则，但是，内网nat环回就不能用了。
> > [image]
> 
> 情况还有点不一样
> 我本来还有一些配置无法全部激活，查看了一次以后发现
> 
> _配置重定向 选项 src_ip '::' 选项 dest_port '3389'_
> 
> src_ip这个地方不正确，手工改成0.0.0.0之后就好了，现在评论正常。
> 
> 最新613这个版本我碰到好几个问题。
> 比如重启之后，客户端无法正常DNS解析，再次重启又好了。
> 又比如经常发现无法修改配置，实质是文件系统只读。
> 能力有效，无法解决这些问题，我现在不恢复320版本了，知道是不是会稳定一点。
[image]
我的文件，原来没有您说的这一条，然后新增了之后，刚开始不行，但是过一段时间就可以了，但是防火墙那里还是必须要设置转发。目前别的问题没有发现，因为我就用，去广告，扶墙，还有WIFI。以及后边客户端的端口转发，现在基本都没啥问题。0613版本。

kidd31, 2021-07-30:
同样是0613版本，也是碰到了同样的问题
不知道跟设置转发的条数有没有关系？我发现好像前十条左右时没问题的，后面的就不行，但是内网输入ddns+端口的形式是可以打开的，说明环回ok，就是外网访问打不开

leo783x, 2021-11-16:
> 那是因为你连接着这些端口，不要连接这些端口，比如，3389 ，先不要远程。然后保存，就能成功。这个应该是BUG，我已经弃用这个了，用其他端口转发工具。

请问用的啥？参考一下。我有十几条转发，设置太累了

closed 2026-10-09 by xiaoqingfengATGH