xiaoqingfengATGH/HomeLede #108: 重启 passwall 导致 dnsmasq 返回 "REFUSED"

tylinux, 2021-06-03:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：重启 passwall 会勾上 dnsmasq 配置里的『忽略解析文件』选项，导致 DNS查询返回 "REFUSED" 状态。

### (2) 路由器型号和固件版本

A：HomeLede v2021.04.18

xiaoqingfengATGH, 2021-06-04:
默认确实不希望dnsmasq使用解析文件。而是希望它使用上游的chinadns-ng。

如果开启解析文件，一个网址请求会同时发给上游及解析文件中的dns，这样哪个返回速度快就会采用哪个地址，很有可能会返回不是你预期的结果。

tylinux, 2021-06-04:
> 默认确实不希望dnsmasq使用解析文件。而是希望它使用上游的chinadns-ng。
> 
> 如果开启解析文件，一个网址请求会同时发给上游及解析文件中的dns，这样哪个返回速度快就会采用哪个地址，很有可能会返回不是你预期的结果。

那我再确认一下到底是谁 REFUSED 了 DNS 请求

huxim, 2021-06-09:
这应该是个feature，重启过程中如果响应请求，返回的就是被污染的结果

tylinux, 2021-06-10:
> 这应该是个feature，重启过程中如果响应请求，返回的就是被污染的结果

嗯嗯，理解了。不过还是没能找到为什么refuse 了DNS请求。上游的 ChinaDNS 都是正常解析，到dnamasq 就gg了

tylinux, 2021-06-10:
可能是我配置的问题，先关闭了，辛苦~

closed 2021-06-10 by tylinux