xiaoqingfengATGH/HomeLede #114: Homeconnect 无法连接

Aldolase, 2021-07-04:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：homeconnect无法连接。做了如下操作
开启了homeconnect ,修改了psk与共享密码，删除了默认的homlede 用户，新建了自己的用户名和密码
android手机无法用L2TP/PSK连接
从网络->防火墙->通信规则里面，打开了500(upd), 1701（udp), 4500(upd)端口，依然无法连接

### (2) 路由器型号和固件版本

A：j1900软路由, 20210613固件版本

### (3) 详细日志

A：找不到具体的日志在哪里，在开启homeconnect后，系统日志中有着一段，也许有用

Sun Jul  4 14:30:04 2021 daemon.err uhttpd[3726]: /usr/share/homeconnect/vpnrpc.js:3672
Sun Jul  4 14:30:04 2021 daemon.err uhttpd[3726]:                 throw err;
Sun Jul  4 14:30:04 2021 daemon.err uhttpd[3726]:                 ^
Sun Jul  4 14:30:04 2021 daemon.err uhttpd[3726]:
Sun Jul  4 14:30:04 2021 daemon.err uhttpd[3726]: Error: connect ECONNREFUSED 127.0.0.1:5555
Sun Jul  4 14:30:04 2021 daemon.err uhttpd[3726]:     at TCPConnectWrap.afterConnect [as oncomplete] (net.js:1135:16) {
Sun Jul  4 14:30:04 2021 daemon.err uhttpd[3726]:   errno: 'ECONNREFUSED',
Sun Jul  4 14:30:04 2021 daemon.err uhttpd[3726]:   code: 'ECONNREFUSED',
Sun Jul  4 14:30:04 2021 daemon.err uhttpd[3726]:   syscall: 'connect',
Sun Jul  4 14:30:04 2021 daemon.err uhttpd[3726]:   address: '127.0.0.1',
Sun Jul  4 14:30:04 2021 daemon.err uhttpd[3726]:   port: 5555
Sun Jul  4 14:30:04 2021 daemon.err uhttpd[3726]: }
Sun Jul  4 14:30:04 2021 daemon.err uhttpd[3726]: /usr/share/homeconnect/vpnclientAjax.js:62
Sun Jul  4 14:30:04 2021 daemon.err uhttpd[3726]: 				let result = {succ:false,msg:handleError(err)}
Sun Jul  4 14:30:04 2021 daemon.err uhttpd[3726]: 				                                         ^
Sun Jul  4 14:30:04 2021 daemon.err uhttpd[3726]:
Sun Jul  4 14:30:04 2021 daemon.err uhttpd[3726]: ReferenceError: err is not defined
Sun Jul  4 14:30:04 2021 daemon.err uhttpd[3726]:     at ClientRequest.<anonymous> (/usr/share/homeconnect/vpnclientAjax.js:62:46)
Sun Jul  4 14:30:04 2021 daemon.err uhttpd[3726]:     at ClientRequest.emit (events.js:315:20)
Sun Jul  4 14:30:04 2021 daemon.err uhttpd[3726]:     at TLSSocket.socketErrorListener (_http_client.js:426:9)
Sun Jul  4 14:30:04 2021 daemon.err uhttpd[3726]:     at TLSSocket.emit (events.js:315:20)
Sun Jul  4 14:30:04 2021 daemon.err uhttpd[3726]:     at emitErrorNT (internal/streams/destroy.js:92:8)
Sun Jul  4 14:30:04 2021 daemon.err uhttpd[3726]:     at emitErrorAndCloseNT (internal/streams/destroy.js:60:3)
Sun Jul  4 14:30:04 2021 daemon.err uhttpd[3726]:     at processTicksAndRejections (internal/process/task_queues.js:84:21)

xiaoqingfengATGH, 2022-05-06:
你似乎修改了softethervpn默认配置？

请不要修改内置配置，另外softethervpn需要DHCP，请保证softethervpn虚拟网卡和LAN桥接，且DHCP工作正常

Aldolase, 2022-05-23:
> 

非常感谢回复，好久了。除了修改用户名和密码外没有做任何修改。现在已经想别的办法回家了。依旧非常感谢

closed 2022-05-23 by Aldolase