xiaoqingfengATGH/HomeLede #116: 请教一下编译时出现softethervpn5报错，请教如何处理？

chinacat911, 2021-07-15:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [X ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题
请教一下编译时出现softethervpn5报错，请教如何处理？
A：

### (2) 路由器型号和固件版本

A：

### (3) 详细日志
 make[3] -C feeds/xiaoqingfeng/softethervpn5 host-compile
    ERROR: package/feeds/xiaoqingfeng/softethervpn5 [host] failed to build.
make: *** [world] Error 1
make -r world: build failed. Please re-run make with -j1 V=s or V=sc for a higher verbosity level to see what's going on
/workdir/openwrt/include/toplevel.mk:228: recipe for target 'world' failed

A：

alexcdever, 2021-07-27:
> Please re-run make with -j1 V=s or V=sc for a higher verbosity level to see what's going on
参照这里你编译的时候带上这两个拿到更详细的记录看看是什么问题，我现在也是在编译到这个失败了，但是搜不到解决方案

xiaoqingfengATGH, 2022-03-31:
之前编译并未遇到过，新版本编译看看是否通过。
可尝试编译前执行一下make dirclean

closed 2022-03-31 by xiaoqingfengATGH