xiaoqingfengATGH/HomeLede #131: Clash订阅成功 还是上不了梯

LilithBristol, 2022-01-04:
## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [✓] 没有类似的issue

## 2. 详细叙述

HomeClash 配置了机场提供的订阅文件
但还是上不去谷歌等外站

[image]

从clash控制面板上可以看到节点都已经正确订阅了

[image]

在其他机器上用clash客户端可以正常连上外站 说明订阅是没问题的

### (1) 具体问题

A：
openwrt ip 192.168.10.60
宿主机 ip 192.168.10.40
主路由 ip 192.168.10.30
均可正常连接互联网
openwrt 和 宿主机网络桥接 可互通

homeclash订阅机场提供的clash配置 订阅成功
节点在clash控制面板上看到

openwrt 全局设置 fake-ip还是redir-host 均无效
未启用任何第三方规则 
选项基本默认

### (2) 路由器型号和固件版本

A：HomeLede v2021.11.10 based on OpenWrt R21.11.11 / LuCI Master (git-21.305.50167-1150dbe)

### (3) 详细日志

日志感觉看不出来啥
```
2022-01-05 18:46:10 警告：OpenClash 启动成功，检测到您启用了IPv6的DHCP服务，可能会造成连接异常！
2022-01-05 18:46:10 第十步: 添加计划任务,启动进程守护程序...
2022-01-05 18:46:09 第九步: 重启 Dnsmasq 程序...
2022-01-05 18:46:09 第八步: 设置防火墙规则...
2022-01-05 18:46:09 第七步: 设置控制面板...
2022-01-05 18:46:09 第六步: 等待主程序下载外部文件...
2022-01-05 18:46:06 第五步: 检查内核启动状态...
2022-01-05 18:46:06 提示: 未检测到特殊配置，调用 Dev 内核启动...
2022-01-05 18:46:06 第四步: 启动主程序...
```

xiaole51, 2022-01-24:
Clash 的问题还真不少. 
我的也是虚拟机+旁路由.
测试俩个版本:
HomeLede v2021.02.27 based on OpenWrt R21.2.1 / LuCI Master (git-21.037.51918-08b66fb)
HomeLede v2021.11.10 based on OpenWrt R21.11.11 / LuCI Master (git-21.305.50167-1150dbe)

ws传输协议的节点全部连不上, 裸奔的tcp协议可以.
同样的节点在passwell一切正常.

xiaole51, 2022-01-24:
误杀了. 
仔细观察配置文件,转换出来的ws节点中缺少skip-cert-verify: true参数导致节点无法连接. 
手动添加此参数后可正常链接.

closed 2022-03-29 by LilithBristol