xiaoqingfengATGH/HomeLede #141: AdguardHome不返回来自IPv6客户端的DNS请求

Houiin, 2022-05-23:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A： 内网IPv6设备向HomeLede请求DNS时提示结果超时。经过测试发现是AdguardHome的问题导致的。

HomeLede获取到到的内网ipv6地址为：`fdb0:d130:f81c:4::1`，可以被其他设备正常访问。

Adguard Home, dnsmasq都开启了IPV6解析

```shell
# 53端口为adguard home 时，DNS请求超时
$ nslookup  WWW.BAIDU.COM fdb0:d130:f81c:4::1
;; connection timed out; no servers could be reached
```
把53端口重定向到`Dnsmasq: 54`后一切正常

```shell
# 设置防火墙规则: 将53重定向到dnsmasq
ip6tables -t nat -A PREROUTING -p udp --dport 53 -j REDIRECT --to-ports 54
iptables -t nat -A PREROUTING -p udp --dport 53 -j REDIRECT --to-ports 54

# dnsmasq正确返回域名对应的地址
$ nslookup  WWW.BAIDU.COM fdb0:d130:f81c:4::1
Server:		fdb0:d130:f81c:4::1
Address:	fdb0:d130:f81c:4::1#53

Non-authoritative answer:
WWW.BAIDU.COM	canonical name = www.a.shifen.COM.
Name:	www.a.shifen.COM
Address: 36.152.44.96
```

### (2) 路由器型号和固件版本

A： x86  IMG镜像 (Proxmox 7.0.1 KVM虚拟机)
 `HomeLede v2022.05.10 based on OpenWrt R22.5.5 / LuCI Master (git-22.125.21429-b7daf4c)`

### (3) 详细日志

```
Mon May 23 10:33:06 2022 daemon.notice netifd: Network alias 'br-lan' link is up
Mon May 23 10:33:06 2022 daemon.notice netifd: Interface 'lan6' has link connectivity
Mon May 23 10:33:06 2022 daemon.notice netifd: Interface 'lan6' is enabled
Mon May 23 10:33:06 2022 daemon.notice netifd: Interface 'lan6' is setting up now
Mon May 23 10:33:08 2022 daemon.notice netifd: Interface 'lan6' is now up
Mon May 23 10:33:08 2022 user.warn mwan3-hotplug[11028]: hotplug called on lan6 before mwan3 has been set up
Mon May 23 10:33:09 2022 user.notice nlbwmon: Reloading nlbwmon due to ifup of lan6 (br-lan)
Mon May 23 10:33:09 2022 daemon.err nlbwmon[11494]: The netlink receive buffer size of 524288 bytes will be capped to 212992 bytes
Mon May 23 10:33:09 2022 daemon.err nlbwmon[11494]: by the kernel. The net.core.rmem_max sysctl limit needs to be raised to
Mon May 23 10:33:09 2022 daemon.err nlbwmon[11494]: at least 524288 in order to sucessfully set the desired receive buffer size!
Mon May 23 11:11:55 2022 daemon.info netdata[5766]: Initializing file /var/cache/netdata/cpu.cpu0_softirqs/HRTIMER.db.
Mon May 23 11:35:47 2022 user.info : Restarting firewall on custom /etc/firewall.user change
Mon May 23 11:35:59 2022 user.info : Restarting firewall on custom /etc/firewall.user change
Mon May 23 12:00:00 2022 cron.err crond[6601]: USER root pid 27685 cmd [ -x /etc/init.d/smartdns ] && /etc/init.d/smartdns restart > /dev/null 2>&1 &
Mon May 23 13:42:48 2022 user.info : Restarting firewall on custom /etc/firewall.user change
Mon May 23 13:43:16 2022 daemon.notice ttyd[6765]: HTTP / - 192.168.5.230
Mon May 23 13:43:16 2022 daemon.notice ttyd[6765]: HTTP /token - 192.168.5.230
Mon May 23 13:43:16 2022 daemon.notice ttyd[6765]: WS   /ws - 192.168.5.230, clients: 1
Mon May 23 13:43:16 2022 daemon.notice ttyd[6765]: started process, pid: 14256
Mon May 23 13:43:20 2022 auth.info login[14267]: root login on 'pts/0'
Mon May 23 13:44:10 2022 user.info : Restarting firewall on custom /etc/firewall.user change
Mon May 23 13:47:20 2022 user.info : Restarting firewall on custom /etc/firewall.user change
Mon May 23 13:47:54 2022 user.info : Restarting firewall on custom /etc/firewall.user change
Mon May 23 13:48:39 2022 user.info : Restarting firewall on custom /etc/firewall.user change
```

Houiin, 2022-05-24:
如果ipv6客户端也要走adg的话，就只能把adg放在dnsmasq后面了，缺点就是看不见来源了，只有127.0.0.1

SakuraFallingMad, 2022-06-05:
自行添加适配的adg的设置到adg的yaml文件

OpportunityLiu, 2022-10-16:
[image]
这里，我记得默认写的是所有 IPv4 的内网 CIDR，直接清空，允许所有客户端就行了

closed 2023-02-08 by Houiin