xiaoqingfengATGH/HomeLede #157: 海外分流解析会停用

vbskycn, 2023-08-06:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：我是用爱快主路，开dhcp,op旁边路，关dhcp,试了很多遍，新安装的固件，只要把op的IP改了，海外分流解析就会停用
[image]

### (2) 路由器型号和固件版本

A：pve下的x86 2023.2.5版

### (3) 详细日志

A：太长了发不出来

zurica2013, 2023-08-20:
[image]
完全一样的问题，试过重启和修改各种配置都不生效。刚想提问，发现一样的情况：
```
root@OpenWrt:~# nslookup www.google.com 127.0.0.1:5053
nslookup: write to '127.0.0.1:5053': Connection refused
;; connection timed out; no servers could be reached
```
另外

samjoy1005, 2023-12-03:
一样的情况

closed 2026-10-09 by xiaoqingfengATGH